BillyBlaze/OctoPrint-TouchUI #231: Temperature-Graph too small on 7" Touchdisplay on 90° Rotated

DominikJosh, 2018-08-09:
Hello,

can I made the Temp-Graph bigger? The Problem is that it is very small on a Display what is Horizontal, when the Display rotate to vertical it is good. So what can I do that the Temp-Graph would be bigger. Sry for the bad english ^^

Greetings Dominik

CTcartesian3D, 2018-08-10:
Does it look like this...

[image]

CTcartesian3D, 2018-08-11:
Yep, same issue I've been having...

Take a look at this thread on octoprint.org

[TouchUI Temperature-graph Squished ~ Modify size](https://discourse.octoprint.org/t/touchui-temperature-graph-squished-modify-size/3014/49)

And this #226

BillyBlaze, 2018-09-01:
Thanks for the request. I will address this issue in the upcoming release!

closed 2018-09-15 by BillyBlaze

BillyBlaze, 2018-09-15:
This has been released in 0.3.12

commit 81ea4c07 (2018-09-01): :art: Improve tempgraph responsiveness

Fixes #231 and #226