BillyBlaze/OctoPrint-TouchUI #232: Control interface not showing

MisterFoxi, 2018-08-11:
Hi,
I just installed Touchui using a 3.5 adafruit display on a Pi 3. All works except the control tab (3rd from left) display nothing at all. 

Is there some kind of config to add ?

Thanks

Laurent

BillyBlaze, 2018-09-02:
Hi @MisterFoxi 

Are you logged in as an user? If you're not logged in it will not display anything.

MisterFoxi, 2018-09-08:
That's the issue. The login area shows partially on the 3.5 screen and there is no way to scroll that part. If you try to scroll, it closes the login UI automatically.

BillyBlaze, 2018-09-12:
It should close while scrolling.

You can uninstall TouchUI and install it again with the following URL: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

I think that should fix it, if not, please provide me a list of installed plugins so I can try to reproduce your issue.

closed 2018-10-03 by stale[bot]