BillyBlaze/OctoPrint-TouchUI #233: Touch UI will not start

cobra281, 2018-08-15:
I followed the directions to the letter at https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(Jessie-Full)

The Pi will boot to a CLI

When entering sudo xinit ~/TouchUI-autostart/chromium.xinit to try to manually start TouchUI I get the follwing error:

xterm: No absolute path found for shell: ~/TouchUI-autostart/chromium.xinit
xinit: connection to X server lost

Ideas

BillyBlaze, 2018-09-02:
Hi @cobra281,

What happens if you try `sudo xinit /home/pi/TouchUI-autostart/chromium.xinit` ?

closed 2018-09-23 by stale[bot]

ricardohalfeld, 2020-02-23:
I had the same issue just now. 

> Hi @cobra281,
> 
> What happens if you try `sudo xinit /home/pi/TouchUI-autostart/chromium.xinit` ?

Does the trick, although my different hostname likely caused it to not find the page.

ricardohalfeld, 2020-02-23:
I noticed some of the scripts assume pi as the user, and I created another before trying the whole TouchUI thing. Maybe the key is introducing the variable $USER in the scripts.