BillyBlaze/OctoPrint-TouchUI #234: hang at "Loading TouchUI"

Ch3fk0ch88, 2018-08-19:
Hello,
i installed everything according to the instructions on github.
Unfortunately the display hangs at "Loading TouchUI" and I don't know why.

Can anyone help me?

If you need any information, please tell me which and how I can get it.

Thanks.

bearink, 2018-08-20:
I am having the same issue, reinstalled fresh and tried multiple fixes around here.

first it connects then jump on to Loading TouchUI for a while before a fail
then after touching the screen, it returns to loading TouchUI

smiler, 2018-08-23:
I have the same issue. New install following https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

smiler, 2018-08-23:
Most likely a duplicate of #228

redwolf3773, 2018-08-29:
I am having the same issue as well. It hangs up at Loading Touch UI. This is not a new install. I did update Octopi to the latest version and it seemed to happen then.

bearink, 2018-08-29:
I did reinstall the whole thing again without updating octopi this time and everything works as it should.

BillyBlaze, 2018-09-01:
Thanks everyone for the feedback, I am currently working on this issue and plan to release a.s.a.p.

However this issue is a similar to #221 so therefor I am closing this.

closed 2018-09-01 by BillyBlaze