BillyBlaze/OctoPrint-TouchUI #235: No possibility to get it running

dudljump, 2018-08-31:
After installing a fresh image and configuring all things right as it is written in the Instructions there is only a black Screen after booting up (Waveshare 3.5 LCD Touch-Display)
As I type in the chromium-browser command the following is comming out:

pi@octopi:~ $ chromium-browser
[2407:2407:0831/150702.511726:ERROR:browser_main_loop.cc(670)] Failed to put Xlib into threaded mode.

(chromium-browser:2407): Gtk-WARNING **: cannot open display:
pi@octopi:~ $

Is there any solution?

Running on Octopi and Octoprint Version 1.38

BillyBlaze, 2018-09-02:
Hi @dudljump,

Could you perhaps share the contents of `/var/log/Xorg.0.log`?

StAnLi89, 2018-09-03:
Same error code (670)

Log content:
pi@octopi:~ $ sudo nano /var/log/Xorg.0.log
[sudo] Passwort für pi:
  GNU nano 2.7.4                                                                                                                                    Datei: /var/log/Xorg.0.log                                                                                                                                    Verändert

        built-ins
[     7.103] (==) ModulePath set to "/usr/lib/xorg/modules"
[     7.103] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[     7.112] (II) Loader magic: 0x1fbf40
[     7.113] (II) Module ABI versions:
[     7.113]    X.Org ANSI C Emulation: 0.4
[     7.113]    X.Org Video Driver: 23.0
[     7.113]    X.Org XInput driver : 24.1
[     7.113]    X.Org Server Extension : 10.0
[     7.117] (--) using VT number 2

[     7.117] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[     7.118] (II) no primary bus or device found
[     7.126] (II) LoadModule: "glx"
[     7.142] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     7.330] (II) Module glx: vendor="X.Org Foundation"
[     7.330]    compiled for 1.19.2, module version = 1.0.0
[     7.330]    ABI class: X.Org Server Extension, version 10.0
[     7.330] (II) LoadModule: "fbdev"
[     7.331] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[     7.334] (II) Module fbdev: vendor="X.Org Foundation"
[     7.334]    compiled for 1.19.0, module version = 0.4.4
[     7.334]    Module class: X.Org Video Driver
[     7.334]    ABI class: X.Org Video Driver, version 23.0
[     7.334] (II) FBDEV: driver for framebuffer: fbdev
[     7.342] (WW) Falling back to old probe method for fbdev
[     7.342] (II) Loading sub module "fbdevhw"
[     7.342] (II) LoadModule: "fbdevhw"
[     7.343] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[     7.353] (II) Module fbdevhw: vendor="X.Org Foundation"
[     7.353]    compiled for 1.19.2, module version = 0.0.2
[     7.353]    ABI class: X.Org Video Driver, version 23.0
[     7.354] (II) FBDEV(0): using /dev/fb0
[     7.354] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[     7.354] (II) FBDEV(0): Creating default Display subsection in Screen section
        "Default Screen Section" for depth/fbbpp 24/32
[     7.354] (==) FBDEV(0): Depth 24, (==) framebuffer bpp 32
[     7.354] (==) FBDEV(0): RGB weight 888
[     7.354] (==) FBDEV(0): Default visual is TrueColor
[     7.354] (==) FBDEV(0): Using gamma correction (1.0, 1.0, 1.0)
[     7.354] (II) FBDEV(0): hardware: BCM2708 FB (video memory: 1500kB)
[     7.354] (**) FBDEV(0): Option "fbdev" "/dev/fb0"
[     7.354] (II) FBDEV(0): checking modes against framebuffer device...
[     7.354] (II) FBDEV(0): checking modes against monitor...
[     7.355] (--) FBDEV(0): Virtual size is 800x480 (pitch 800)
[     7.355] (**) FBDEV(0):  Built-in mode "current"
[     7.355] (==) FBDEV(0): DPI set to (96, 96)
[     7.355] (II) Loading sub module "fb"
[     7.355] (II) LoadModule: "fb"
[     7.356] (II) Loading /usr/lib/xorg/modules/libfb.so
[     7.366] (II) Module fb: vendor="X.Org Foundation"
[     7.366]    compiled for 1.19.2, module version = 1.0.0
[     7.366]    ABI class: X.Org ANSI C Emulation, version 0.4
[     7.366] (**) FBDEV(0): using shadow framebuffer
[     7.366] (II) Loading sub module "shadow"
[     7.366] (II) LoadModule: "shadow"
[     7.366] (II) Loading /usr/lib/xorg/modules/libshadow.so
[     7.376] (II) Module shadow: vendor="X.Org Foundation"
[     7.377]    compiled for 1.19.2, module version = 1.1.0
[     7.377]    ABI class: X.Org ANSI C Emulation, version 0.4
[     7.377] (==) Depth 24 pixmap format is 32 bpp
[     7.406] (==) FBDEV(0): Backing store enabled
[     7.419] (==) FBDEV(0): DPMS enabled
[     7.424] (==) RandR enabled
[     7.479] (II) SELinux: Disabled on system
[     7.488] (II) AIGLX: Screen 0 is not DRI2 capable
[     7.488] (EE) AIGLX: reverting to software rendering
[     7.489] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so: cannot open shared object file: No such file or directory)
[     7.489] (EE) GLX: could not load software renderer
[     7.489] (II) GLX: no usable GL providers found for screen 0
[     7.770] (II) config/udev: Adding input device Logitech K400 Plus (/dev/input/event1)
[     7.770] (II) No input driver specified, ignoring this device.
[     7.770] (II) This device may have been added with another device file.
[     7.773] (II) config/udev: Adding input device Logitech K400 Plus (/dev/input/mouse1)
[     7.773] (II) No input driver specified, ignoring this device.
[     7.773] (II) This device may have been added with another device file.
[     7.774] (II) config/udev: Adding input device FT5406 memory based driver (/dev/input/event0)
[     7.774] (II) No input driver specified, ignoring this device.
[     7.774] (II) This device may have been added with another device file.
[     7.775] (II) config/udev: Adding input device FT5406 memory based driver (/dev/input/mouse0)
[     7.775] (II) No input driver specified, ignoring this device.
[     7.775] (II) This device may have been added with another device file.

BillyBlaze, 2018-09-03:
```
pi@octopi:~ $ chromium-browser
[2658:2658:0903/202740.537909:ERROR:browser_main_loop.cc(670)] Failed to put Xlib into threaded mode.

(chromium-browser:2658): Gtk-WARNING **: cannot open display:
```

This error can be ignored, the only reason we are issuing that command is to see if Chromium is compatible with your kernel. If it tries to open the display then it is compatible.

The other log also doesn't indicate any problems.

What happens if you try this:
1. `sudo service touchui stop`
1. `sudo xinit ~/TouchUI-autostart/chromium.xinit`

otherguy, 2018-09-03:
I have the same issue. I run the `xinit` command via SSH and want TouchUI to show up on an official Raspberry Touchscreen.

```$ sudo xinit ~/TouchUI-autostart/chromium.xinit
X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=37d79269-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait logo.nologo
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Current version of pixman: 0.34.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Mon Sep  3 22:31:29 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
TouchUI: OctoPi detected
TouchUI: touch events enabled
TouchUI: screensaver enabled
wm_set_cursor_visibility: visible using XCreateFontCursor
Fontconfig warning: "/etc/fonts/fonts.conf", line 160: blank doesn't take any effect anymore. please remove it from your fonts.conf
[1960:1960:0903/223132.688735:ERROR:gl_surface_glx.cc(423)] glxQueryVersion failed
[1960:1960:0903/223132.689167:ERROR:gl_initializer_x11.cc(157)] GLSurfaceGLX::InitializeOneOff failed.
[1960:1960:0903/223132.695837:ERROR:viz_main_impl.cc(197)] Exiting GPU process due to errors during initialization
[1869:1904:0903/223132.700502:ERROR:service_manager_context.cc(252)] Attempting to run unsupported native service: /usr/lib/chromium-browser/content_gpu.service
[1869:1904:0903/223132.708809:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to launch GPU process.
[1869:2043:0903/223132.808497:ERROR:bus.cc(394)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1869:1904:0903/223133.313549:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to launch GPU process.
[1869:1869:0903/223133.352977:ERROR:gpu_process_transport_factory.cc(1029)] Lost UI shared context.```

TouchUI starts but after some time I get "Startup failed" on the external touchscreen.

otherguy, 2018-09-03:
On second thought it's probably a different issue since @dudljump gets no output at all 🤔

BillyBlaze, 2018-09-03:
Yeah, you're probably affected by #221 (which is currently fixed on the maintenance branch)

otherguy, 2018-09-03:
@BillyBlaze awesome! Thanks for the quick answer! That was indeed the issue 🙂 Sorry for posting in the wrong thread!

StAnLi89, 2018-09-04:
On startup i get the TouchUI Loading screen, but after some time it shows: "Startup failed... Tap to retry" the Touchscreen doesn't respond though

Log on trying restarting the service as you mentioned brought up this in the shell:

X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 ar                                mv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_                                fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 cons                                ole=ttyS0,115200 console=tty1 root=PARTUUID=544f6a12-02 rootfstype=ext4 elevator=deadline fs                                ck.repair=yes rootwait
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Sep  4 16:59:34 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
TouchUI: OctoPi detected
TouchUI: touch events automatically detected
TouchUI: screensaver enabled
wm_set_cursor_visibility: visible using XCreateFontCursor
Fontconfig warning: "/etc/fonts/fonts.conf", line 160: blank doesn't take any effect anymore                                . please remove it from your fonts.conf
[2070:2070:0904/165936.373205:ERROR:gl_surface_glx.cc(423)] glxQueryVersion failed
[2070:2070:0904/165936.373377:ERROR:gl_initializer_x11.cc(157)] GLSurfaceGLX::InitializeOneO                                ff failed.
[2070:2070:0904/165936.385537:ERROR:viz_main_impl.cc(197)] Exiting GPU process due to errors                                 during initialization
[1961:2009:0904/165936.399992:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to laun                                ch GPU process.
[1961:2199:0904/165936.585140:ERROR:bus.cc(394)] Failed to connect to the bus: Could not par                                se server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix                                ")
[1961:2009:0904/165936.826170:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to laun                                ch GPU process.
[1961:1961:0904/165936.857472:ERROR:gpu_process_transport_factory.cc(1029)] Lost UI shared c                                ontext.

alex-mohemian, 2018-09-04:
@StAnLi89 did you try the solution in 
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/221 ?

otherguy, 2018-09-04:
@StAnLi89 did you try the solution in https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/221?

StAnLi89, 2018-09-04:
Didnt see that one sorry, ill try it now.

StAnLi89, 2018-09-04:
Worked for me, it stats now.
Now trying to get the Touchscreen to respond xD

dudljump, 2018-09-04:
Sorry for the late answer - my raspi crashed totally so that I have to buy a new one... Long story. But for now there is no longer a issue from my site

BillyBlaze, 2018-09-04:
@dudljump sad to hear that, good luck!

@otherguy thanks for pointing in the right direction

@StAnLi89 open a new issue for other issues

closed 2018-09-04 by BillyBlaze