BillyBlaze/OctoPrint-TouchUI #237: Calibration Error

eboston, 2018-09-09:
Following the directions for calibration here...
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Troubleshooting#calibration

Step 2 is to run _curl -sLS https://apt.adafruit.com/add | sudo bash_ but it seems the web site is no longer available.

BillyBlaze, 2018-09-09:
Just skip that step, seems like xinput-calibration is listed on the default repository of Stretch.

BillyBlaze, 2018-09-09:
1. Use `/etc/X11/xorg.conf.d/` instead of `/etc/X11/xorg.conf.d/`
    (/etc/X11/xorg.conf.d/ isn't generated anymore by default)

2. Yeah your right, feel free to edit the public wiki!

eboston, 2018-09-09:
Use the directory that isn't generated anymore by default?  :) You have 
the same directory in all 3 references.

BillyBlaze, 2018-09-09:
Ouchy, copy-paste mistake. LOL!

I meant use `/usr/share/X11/xorg.conf.d/` instead of `/etc/X11/xorg.conf.d/`

BillyBlaze, 2018-09-10:
Your welcome and thank you for the wiki edits!

closed 2018-09-10 by BillyBlaze