BillyBlaze/OctoPrint-TouchUI #238: Doesn´t autostart on LCD 7Inch HDMI Display

maganzi, 2018-09-12:
Hello,
thanks for your support!

Raspberry Pi 3 Model B
7 Inch 800x480 LCD Display HDMI (B)

i tried everything to autostart TouchUI on this Display but... every time after booting i´am logged in as pi@octopi in CLI. 

1) Clean installation octoprint (2018-06-27-raspbian-stretch) on SD
2) install the 7 Inch Display (https://github.com/goodtft/LCD-show)
3) install TouchUI with Plugin Manager (i also tried OctoPrint-TouchUI-maintenance.zip)
4) install Boot-to-browser OctoPi-Mehtode (https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup)

what i am doing wrong? Can you help me please?

Greetings

maganzi, 2018-09-12:
Hi @BillyBlaze 
thank you, i don´t know but this file is empty
[image]

BillyBlaze, 2018-09-12:
1. Are you sure your SD card is expanded?
1. What is your current config? `sudo cat /usr/share/X11/xorg.conf.d/99-fbdev.conf`

maganzi, 2018-09-12:
1. i expand the filesystem with sudo raspi-config
2.
Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb1"
EndSection
[image]
2. it doesn´t match with my settings

[image]

maganzi, 2018-09-12:
[image]

"/var/log/Xorg.0.log"

BillyBlaze, 2018-09-12:
Ok, you're using the wrong config.

Change
`Option "fbdev" "/dev/fb1"`
into
`Option "fbdev" "/dev/fb0"`

maganzi, 2018-09-12:
It work!! THANK YOU so much!

BillyBlaze, 2018-09-12:
Your welcome! Enjoy!

closed 2018-09-12 by BillyBlaze