BillyBlaze/OctoPrint-TouchUI #239: Bootloader refuses to connect to localhost:80 with OctoPi

BillyBlaze, 2018-09-13:
@jraydoran what was it connecting to? at the bottom of the bootloader it should say something like: 'Connecting to http://localhost/'

Also what is the output of these commands?
`wget 127.0.0.1`
`netstat -tpln` 
`sudo netstat -tulpen | grep haproxy`
`ping 127.0.0.1`

jraydoran, 2018-09-13:
@BillyBlaze, it says "Connecting to http://localhost/"

`pi@octopi:/etc/default $ wget 127.0.0.1`
`--2018-09-13 14:46:20--  http://127.0.0.1/`
`Connecting to 127.0.0.1:80... connected.`
`HTTP request sent, awaiting response... 200 OK`
`Length: 903494 (882K) [text/html]`
`index.html: Permission denied`

`Cannot write to ‘index.html’ (Permission denied).`

`pi@octopi:/etc/default $ sudo netstat -tulpen | grep haproxy`
`[sudo] password for pi:`
`tcp6       0      0 :::80                   :::*                    LISTEN      0          12811      635/haproxy`
`tcp6       0      0 :::443                  :::*                    LISTEN      0          12812      635/haproxy`
`udp        0      0 0.0.0.0:33302           0.0.0.0:*                           0          12813      633/haproxy-master`

`pi@octopi:/etc/default $ ping 127.0.0.1`
`PING 127.0.0.1 (127.0.0.1) 56(84) bytes of data.`
`64 bytes from 127.0.0.1: icmp_seq=1 ttl=64 time=0.127 ms`
`64 bytes from 127.0.0.1: icmp_seq=2 ttl=64 time=0.092 ms`
`64 bytes from 127.0.0.1: icmp_seq=3 ttl=64 time=0.111 ms`
`64 bytes from 127.0.0.1: icmp_seq=4 ttl=64 time=0.096 ms`
`64 bytes from 127.0.0.1: icmp_seq=5 ttl=64 time=0.106 ms`
`64 bytes from 127.0.0.1: icmp_seq=6 ttl=64 time=0.105 ms`

I hope this is readable. 

Thank you!

BillyBlaze, 2018-09-13:
ermm, `index.html: Permission denied`

What is your Haproxy config? (`cat /etc/haproxy/haproxy.cfg`)
Does `http://octopi.local/` work for you, or do you connect differently from a computer inside your network?

BillyBlaze, 2018-09-13:
or wait, it's listening to ipv6 only???

What is the output of:
`curl -g "http://[::1]/"`

jraydoran, 2018-09-13:
`curl -g "http://[::1]/"`

returns with pages of http code. Looks good.

BillyBlaze, 2018-09-13:
awesome, thanks for confirming, can you still paste your Haproxy config? So I can reproduce the issue and verify myself that it has been fixed?

jraydoran, 2018-09-13:
As requested.

pi@octopi:~ $ cat /etc/haproxy/haproxy.cfg
global
        maxconn 4096
        user haproxy
        group haproxy
        log 127.0.0.1 local1 debug

defaults
        log     global
        mode    http
        option  httplog
        option  dontlognull
        retries 3
        option redispatch
        option http-server-close
        option forwardfor
        maxconn 2000
        timeout connect 5s
        timeout client  15min
        timeout server  15min

frontend public
        bind :::80 v4v6
        bind :::443 v4v6 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint

backend octoprint
        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        reqrep ^([^\ :]*)\ /(.*) \1\ /\2
        reqadd X-Scheme:\ https if needs_scheme { ssl_fc }
        reqadd X-Scheme:\ http if needs_scheme !{ ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend webcam
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http

BillyBlaze, 2018-09-15:
Thanks again for all the info.

Can you update your bootloader once again and check if the latest fix did the job for you?

jraydoran, 2018-09-15:
Thanks for the effort.

I updated the the latest bootloader, but still no luck

It is trying to connect to:
http:localhost:5000

BillyBlaze, 2018-09-15:
Ugh, I was confident that fixed it...

But it seems some OctoPi detection failed.

Can you run this and post the output of it here so I have some idea where it is going wrong.

1. `sudo service touchui stop`
1. `sudo xinit ~/TouchUI-autostart/chromium.xinit`

jraydoran, 2018-09-15:
See below.

pi@octopi:~ $ sudo service touchui stop
pi@octopi:~ $ sudo xinit ~/TouchUI-autostart/chromium.xinit
[sudo] password for pi:

X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.14.68-v7+ #1140 SMP Wed Sep 5 17:17:58 BST 2018 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=765647d0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sat Sep 15 20:36:36 2018
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
TouchUI: OctoPi detected
TouchUI: haproxy has a listening IPv6 socket.
TouchUI: touch events enabled
TouchUI: screensaver enabled
wm_set_cursor_visibility: visible using XCreateFontCursor
Fontconfig warning: "/etc/fonts/fonts.conf", line 160: blank doesn't take any effect anymore. please remove it from your fonts.conf
[1586:1586:0915/203639.676943:ERROR:gl_surface_glx.cc(423)] glxQueryVersion failed
[1586:1586:0915/203639.677086:ERROR:gl_initializer_x11.cc(157)] GLSurfaceGLX::InitializeOneOff failed.
[1586:1586:0915/203639.686092:ERROR:viz_main_impl.cc(197)] Exiting GPU process due to errors during initialization
[1488:1523:0915/203639.701431:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to launch GPU process.
[1488:1713:0915/203639.884992:ERROR:bus.cc(394)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1488:1523:0915/203640.190535:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to launch GPU process.
[1488:1488:0915/203640.215756:ERROR:gpu_process_transport_factory.cc(1029)] Lost UI shared context.

[touchui.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/2385996/touchui.txt)

BillyBlaze, 2018-09-15:
Hmm detection is going fine... maybe the javascript file has been cached?

Clear the cache
`sudo rm -r  ~/.cache/Chromium/Default/Cache/*`

and rerun
`sudo service touchui restart`

BillyBlaze, 2018-09-15:
or...

I just updated the bootloader with an own server and a no-cache header. Can you update your bootloader and check if it now does want to connect to `http://[::1]/` instead

BillyBlaze, 2018-09-16:
Awesome! Your welcome and thank you too for your time and input in resolving these issues! I couldn't have done it without it.

closed 2018-09-16 by BillyBlaze