BillyBlaze/OctoPrint-TouchUI #240: Blinking Logo

sarf2k4, 2018-09-21:
Hi,

I followed the wiki at [here](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)) with a [generic 3.5" tft lcd ](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)). Everything is done, bootloader, updated and all, all I get is the touchui logo kept flashing/blinking without any interface.

I noticed there is small lightning icon on the tft display and it might be the problem where the pi didn't get enough power to start the touchui properly. The thing is, the web interface is working fine except the touchui.

I am using raspberry pi 3 B+

BillyBlaze, 2018-09-28:
Thanks for reporting.

1. What is the message its stuck on?
2. Can you post your octoprint.log'?

sarf2k4, 2018-09-28:
Hi, here is the answer

1. There is no message. There is fade in animation, the moment that "touch UI" is full brightness, it just gone and repeats again.

2. I've attached the octoprint log dated back in 21st september, before I posted this issue. 

[octoprint.log.2018-09-21.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/2427456/octoprint.log.2018-09-21.txt)

Also I tried on a slightly higher powered usb adapter, the results were the same. There is one time however, the screen was black, when I touch the screen, I saw the interface but for short while about 500 millisecond then it was gone to black.

BillyBlaze, 2018-09-28:
It should say "Connecting to TouchUI" or "Loading OctoPrint" also what is the URL it tries to connect to? Maybe you're are affected with the same issue as #242

sarf2k4, 2018-09-28:
I'm sorry, I'm not linux proficient user, I need guide. How do I check the file again on server it will connect for example that #242 issue, to modify the file to localhost but I don't know where it is located to be precise.

The fade in animation is about 500 millisecond, but there is no message I can confirm that. Because the screen just goes black and repeats again. If I'm correct, if there touchui has some problem, it would stay on the touchUI logo with message is it not?

BillyBlaze, 2018-10-28:
Hi @sarf2k4,

Could you [update your bootloader](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader) to the latest version to see if you still have this issue.

pugmad, 2018-10-31:
Same problem for me too I use a Waveshare 3,5 LCD and TouchUI blinking when loading and after is black screen (when i touch it the position i touch is shown for a while) any help would help (i have updated bootloader and nothing changed

closed 2018-11-21 by stale[bot]

Tenember, 2019-03-06:
I'm having the exact same issue.

Screen is black except for an underscore on top left (like a console was showing instead).
Drivers on the 3.5" are working properly (work great in desktop mode).
TOUCH UI is sowing and blinking for a few second (only and only that).
Bootloader is up to date.
Like @pugmad , I can touch the screen and Touchui is shown for half a second, then I can't reproduce it and it goes back to this black screen/console.

ksztibor, 2019-03-15:
The same with me :(

ksztibor, 2019-03-15:
Somehow this helped me regain the GUI Desktop:
https://www.raspberrypi.org/forums/viewtopic.php?t=186996

sarf2k4, 2019-03-17:
> 
> 
> Somehow this helped me regain the GUI Desktop:
> https://www.raspberrypi.org/forums/viewtopic.php?t=186996

Are you able to see the octoprint interface with the solution?
I will have to read thoroughly of the link you posted later on

ksztibor, 2019-03-17:
> > Somehow this helped me regain the GUI Desktop:
> > https://www.raspberrypi.org/forums/viewtopic.php?t=186996
> 
> Are you able to see the octoprint interface with the solution?
> I will have to read thoroughly of the link you posted later on

When I experienced the problem described in this given issue (on the tft there was blinking 'TouchUI' logo, then black screen), in my case the octoprint service itself started mostly, so I could access it from my laptops web browser. But sometimes not even octoprint started. And even now, after the original problem has been solved for me, it occurs, that octoprint doesn't start. Based on my very recent experience, this is caused by the occasional undervoltage of the Pi. I use Pi 3B+, connecting 1 or 2 usb webcams to it. In my case the usb webcams cause this occasional undervoltage, because as soon as I detach them from the Pi, there is no undervoltage: the small lightning icon doesn't appear on the tft.

Concerning the given issue, I'm not sure what exactly did the trick, but in the following list I will describe everything that I did:
1. I detached the usb webcams from the Pi (as described above)
2. In the Octoprint plugins, I uninstalled the Themeify plugin (there is another issue for TouchUI, which described a possible conflict)
3. In the Octoprint plugins, I temporarily disabled the TouchUI plugin.
4. I uninstalled the 'Boot to Browser' function (sudo ~/TouchUI-autostart/helpers/uninstall)
5. I reinstalled the GUI and the tft driver (that probably didn't change anything)
6. Based on https://www.raspberrypi.org/forums/viewtopic.php?t=186996, I tested the framebuffer both with /dev/fb0 and with /dev/fb1. For example, for /dev/fb0, the steps are the following:
a) sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
b) In the line
Option "fbdev" "/dev/fb1"
replace "/dev/fb1" with "/dev/fb0", then save and exit
c) sudo FRAMEBUFFER=/dev/fb1 startx
After that my tft display started to show the actual screen content of the Pi (actually, only on the top half of the screen, but after a reboot, it got fine).
Note: Currently, in /etc/X11/xorg.conf.d/99-fbdev.conf I have "/dev/fb1", but in /usr/share/X11/xorg.conf.d/99-fbturbo.conf I have "/dev/fb0", so maybe it has not much to do with the given issue, but anyhow, the 'sudo FRAMEBUFFER=/dev/fb1 startx' command is a useful tool for testing/forcing the framebuffer.
7. As soon as I had the GUI again, I enabled the TouchUI plugin again, installed the 'Boot to Browser' function again, and now it works fine.