BillyBlaze/OctoPrint-TouchUI #242: Connecting to TouchUI times out.

andrewsil1, 2018-09-27:
Raspberry Pi 3, Jessie, and Octoprint, all up to date.  Using the Adafruit 3.5” resistive touchscreen.

Device boots, shows the touch cursor, then the TouchUI boot screen.  It counts to 30 and eventually says Connecting to Touch UI failed.  At the bottom it says “Connecting to http://[::1]/“
I can poke the screen and it tries again, but still fails to connect.

I’ve also uninstalled the release version and installed the latest maintenance drop of TouchUI instead, but this didn’t change the behavior.

TouchUI works fine from an external browser, just not from the touch screen.

Suggestions?

Thanks,
Andy

BillyBlaze, 2018-09-28:
Thanks for reporting! I'll have a look into this

bolsoncerrado, 2018-10-06:
Same here...

@andrewsil1  how did u force IPV4?

[image]

bolsoncerrado, 2018-10-06:
Weird, I disabled IPV6 on the whole raspbian and I no longer get the Connecting to http...:1 but I still can't load touchui :???

bolsoncerrado, 2018-10-06:
Hmmmm how weird. I managed to fix it ONLY IF:

1) i globally disable IPv6
2) HARDCODED 		PORT="80" to 		PORT="8056" (I don't use default port 80 on any of my Pi's in the LAN).

The weirdest thing is that we run SEVERAL identical Pi's and this is the first time this happens in this very specific Pi :???? I don't get it!?

bolsoncerrado, 2018-10-06:
How weird is this!

After fixing with the above solution I printed a nice beezel for my screen and pi, hooked it up, edited config.txt on /boot to remove the lcd rotate thing, rebooted and now TouchUI won't connect anymore!?!?

Not sure how much related is this to https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/244 but it's ironic we were pretending similar things with different paths and both affected TouchUI :?

andrewsil1, 2018-10-06:
I just edited the chromium.xinit to ignore the decision it’s making based on its attempt to grep for uses of 80 in the listening ports and just to set the output variables with the IPV4 setup rather than being smart about it.  Quick and dirty fix rather than a “correct” one.

wdani, 2018-10-26:
Are there some new or update how to fix this? i have the same isue. stuck on tuchUI startup. 
I am realy new to that so if you have a fix, tell me how to do pls :-)

BillyBlaze, 2018-10-28:
Thanks for reporting, I have fixed this issue in the bootloader + I have introduced a variable that would disable the auto detect functionality in case someone wants to disable that. If you [update your bootloader](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader) then I hope this problem no longer occurs!

In case you want to disable it, then add `AUTODETECT_PORT=false` to your settings under '/etc/default/touchui'

closed 2018-10-28 by BillyBlaze

wdani, 2018-10-29:
i am a complettly noob for that, so what do i need to do step by step? would be cool to have an small guide.

Thx :)

BillyBlaze, 2018-10-29:
Click on the link 'update your bootloader' in the previous comment for the guide.

Good luck!

wdani, 2018-10-29:
> Click on the link 'update your bootloader' in the previous comment for the guide.
> 
> Good luck!

thx that did help now! what i just had the problem that "sudo service touchui restart"  didnt helped, i did restart octopi, then it did work.

iff7378, 2020-03-05:
I'm getting this same issue after upgrading to Octoprint 1.4.0. I tried the bootloader update link here and got a response that it was already up to date.

TheNore, 2020-03-05:
i used this workaround and it was temporarily fixed

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/368#issuecomment-595036339