BillyBlaze/OctoPrint-TouchUI #245: Touch UI on separate Port

Strice91, 2018-10-05:
Thank you very much for this great plugin. I just installed it and it works flawlessly. 

I just wondered if it would be possible to show the Touch UI only when you access the printer on special additional port. So the raspberry would access e.g. localhost:5001 to show the Touch UI on a small local screen. Every other browser could still reach the normal web interface over port 5000, because the Touch UI is not so comfortable when you want to control the printer from a remote PC.

I hope this is the right way to ask. I would work on a solution myself if I would know where to start.

Thank you very much and best regards!

BillyBlaze, 2018-10-05:
Hi @Strice91,

No, it's not possible to access TouchUI on another port. However, if you don't want to use TouchUI on other devices - except a local touchscreen - then you could [disable the auto-detection](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Configuration) functionality. Your touchscreen should still activate TouchUI regardless of this configuration.

After disabling this, you can still activate TouchUI manually on other devices through the TouchUI modal.

Strice91, 2018-10-09:
Hi!

Thank you! This should do the job. 
Thanks again very much for your work. It is a extreme cool interface.

closed 2018-10-09 by Strice91