BillyBlaze/OctoPrint-TouchUI #246: DISABLE_SCREENSAVER and HDMI display

yet-another-average-joe, 2018-10-17:
Hello,

I am using a HDMI display, and "DISABLE_SCREENSAVER=true" does not work.

Maybe this parameter is for DSI port only ?

BillyBlaze, 2018-10-28:
Hi @yet-another-average-joe 

Thanks for creating an issue and awesome that you where able to solve the issue. Can you perhaps share your solution in case anyone have the same issue?

yet-another-average-joe, 2018-10-28:
Hello BillyBlaze.

I am sorry, but I absolutely do not understand how this "issue" was solved ! It seems to me that it was solved by itself... For sure, I did'nt solve it myself ! Is it really an issue ? I have seen people conplaining, But is this TouchUI-related ?

For the last 2 month, I've been installing, reinstalling, testing, torture testing OctoPrint, TouchUI (and also marlin-conf), and always had the annoying screensaver. (I'm preparing some videos about OctoPrint, TouchUI and marlin-conf, 20+ to be exact). One day, I decided to open this thread, and the next day the screen saver decided to let me alone. It's like when you order some components on Aliexpress or eBay, you inform the vendor you got nothing, and the next day everything is there !

I do not understand and I don't remember, but maybe I did an ap-get update and apt-get upgrade. This makes me think it wasnt an issue with TouchUI, but maybe with the X server.

Yesterday, the screen saver decided to pop up once again. I had to restart the system after a Marlin 2.x bugfix update). And since then, the display is on, without any problem. I could'nt reproduce the issue...

For your information, I install OctoPi, TouchUI and Pixel in an unusual way as far as I know.

1 - OctoPi
2 - TouchUI
3 - Pixel

This way, booting to command line, the system directly starts TouchUI, without the Pixel overhead and the kiosque mode crap. If I need to mess around with Pixel for example to image the SD card with a user friendly interface, I kill OctoPrint and TouchUI services (sudo service xxx stop), and then launch Pixel (sudo startx). (using PuTTY and VNC Viewer)

Could this be related to the Pixel screensaver overwriting something or taking some precedence ?

Feel free to ask me to do some testing. But keep in mind that I am *NOT* a Linux user. I've been coding for DOS & Windows for the last 30 years, Arduino for a few years, but never for Linux or RasPi, and I never learned Python (but I plan to...).

(sorry for my bad english...)

closed 2018-11-18 by stale[bot]

yet-another-average-joe, 2019-02-18:
Hello BillyBlaze,

I post again about this issue. Since the 1st post, I faced the problem many times. And now, I am 100% sure it is related to plugins updates. 

- plugin update : screen saver is activated after OctoPrint restart
- generally, the screen saver keeps quiet after the 1st reboot, but sometimes it needs 1 or 2 more (weird...)

Last plugin I updated and had the problem with : Preheat Button (today)

Workaround : reboot the Pi after a plugin update.