BillyBlaze/OctoPrint-TouchUI #248: Two issues Starting up Pi (new) and touch calibration (old)

DesertFF, 2018-10-23:
I have ran into an issue now. When I turn on my pi is doesn't load up to TouchUI any more is stops and ask me to press enter to continue to. If I press Enter it will load up to TouchUI. I never had to do this before. The message above the press enter to continue reads. Cannot open access to console the root account is locked. See slogan(8) man page for details. How do I correct this. The second issue I have is since day one the touch is off. If I touch the bottom of the screen the pointer is at the top and same if i touch the top of the screen the pointer is at the bottom. I have tried calibrating the touch multiple times with the same outcome.

BillyBlaze, 2018-10-28:
Hi @DesertFF,

1. I am not familiar with this error, but after some googling I think it may be related to FSCK. Could you try the [following instruction](https://raspberrypi.stackexchange.com/a/83500).

2. Please check the instructions of your touchscreen maker for calibration instructions. I cannot keep up with the tons of different calibration steps that differ per maker and that change per OS.

DesertFF, 2018-11-01:
Thank you the startup solution fixed the problem. Now just to still and try and trouble shoot the touch issue which I'm still striking out on.

closed 2018-11-22 by stale[bot]