BillyBlaze/OctoPrint-TouchUI #249: Keyboard Layout

JoeBuster, 2018-10-30:
Hy, I come from Germany and use TouchUI on a 7" display.  The raspberry configuration settings of the keyboard and the language are configured in German (ISO) style.

But the layout of the touchscreen keyboard on TouchUI is English. The BT keyboard, which I used on the raspberry (German keyboard layout), also writes in English style - for example Y <-> Z are reversed.

With TouchUI on my Chrome browser on the desktop PC all keyboard and language settings work in German style.

Is this something that TouchUI can influence?

BillyBlaze, 2018-10-31:
Hi @JoeBuster,

Currently it's not possible to change the layout easily, since it is hard-coded [over here](https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/source/js/components/keyboard.js#L19), Making this configurable is not quiet easily, but I will try to add this in the next release.

JoeBuster, 2018-10-31:
Thank you very much for your answer.

commit c9440013 (2019-05-16): :art: Add translation strings to keyboard

Fixes #249
closed 2019-05-25 by BillyBlaze