BillyBlaze/OctoPrint-TouchUI #251: TouchUI not booting

Nickk888SAMP, 2018-10-31:
Hello! 

I've bought a 3.5" TFT display and is working fine after i installed the drivers. The screen shows the command prompt, after installing TouchUI and installing it with the official install guide from BillyBlaze, it's showing `octopi login: _ `( "_" is blinking), it's blinking for about 5-10 seconds, after that, it's going to a kinda new line as it would try to boot into something, but it's not booting, i've tried to use the 
`xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit` command, and i've got this result:

[Pastebin Code](https://pastebin.com/88BXXXxU)

I've updated and upgraded my Pi to the latest versions, but TouchUI's not booting.

What could be the issue?

I'm using Raspberry Pi 3 B+

Regards.

BillyBlaze, 2018-10-31:
Hi @Nickk888SAMP,

I noticed this error
````
Parse error on line 10 of section InputClass in file /etc/X11/xorg.conf.d/99-calibration.conf
The Section keyword requires a quoted string to follow it.
````

it seems like something went wrong with copy pasting, did you perhaps missed EndSection?

Nickk888SAMP, 2018-10-31:
Well i don't know, what should the command look like?

Nickk888SAMP, 2018-10-31:
It looks like this: [https://i.imgur.com/lPK9zLH.png](https://i.imgur.com/lPK9zLH.png)

Nickk888SAMP, 2018-10-31:
Update 2: So, i've got the server running somehow using this step [https://gist.github.com/alepez/6273dc5220c1c5ec5f3f126e739d58bf](https://gist.github.com/alepez/6273dc5220c1c5ec5f3f126e739d58bf)

There are no errors and it's booting, butt not into OctoPrint TouchUI.
I've tried step 6 to 10, all went well, but it's not booting into TouchUI. Instead it's asking me to log-in.

[Picture](https://i.imgur.com/Amc93XT.jpg)

BillyBlaze, 2018-10-31:
run `sudo raspi-config` and change under `Boot Options` to `boot to command line `

Nickk888SAMP, 2018-10-31:
I've changed under "Boot Option" > "Desktop / CLI" to "Console Autologin" and it works! It's booting into TouchUI ^_^ Thank you! Now i need to calibrate the touchscreen because it's inverted strangely.

danconsole, 2018-11-04:
Hi!

I was having this exact same issue and followed the method suggested. However, instead of logging in to TouchUI as it was supposed to, now I have [this](https://imgur.com/WZSaa22). Does anyone have any idea why?

BillyBlaze, 2018-11-04:
Wait, you guys are using `start_chromium.xinit`? that one is really old. How did you get that?

I would remove `/home/pi/OctoPrint-TouchUI-autostart/` folder and [install the latest](
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)) one by redo-ing step 6 and 7

danconsole, 2018-11-04:
I did that, but without any apparent effect. I'm gonna try formatting the sd card and starting from scratch later. Maybe that helps?

Thanks for the help anyways!

BillyBlaze, 2018-11-04:
if you have updated it, then stop touchui `sudo service touchui stop` and run `sudo xinit ~/TouchUI-autostart/chromium.xinit` and post the output here.

BillyBlaze, 2018-11-04:
thanks, whats the contents of `/etc/X11/xorg.conf.d/99-calibration.conf`

danconsole, 2018-11-04:
Not much stuff: `-bash: /etc/X11/xorg.conf.d/99-calibration.conf: Permission denied`

BillyBlaze, 2018-11-04:
use `sudo`

danconsole, 2018-11-04:
tried it previously, here you go: `pi@octopi:~ $ sudo /etc/X11/xorg.conf.d/99-calibration.conf
[sudo] password for pi:
sudo: /etc/X11/xorg.conf.d/99-calibration.conf: command not found
`

BillyBlaze, 2018-11-04:
try `sudo cat /etc/X11/xorg.conf.d/99-calibration.conf`

danconsole, 2018-11-04:
[Here](https://pastebin.com/98MvfvcG) you go.

BillyBlaze, 2018-11-04:
It looks like the wrong double-quotation character is used in the bottom of the config:

```
Section “Device”
  Identifier “touchscreen”
  Driver “fbdev”
  Option “fbdev” “/dev/fb1”
EndSection
```

replace `“` with `"`

Also since my guide don't have this character, it might be good to inform the guide you where following to adjust this error in their guide.

danconsole, 2018-11-04:
Good eye. Thank you so much! That solved it.

closed 2018-11-25 by stale[bot]