BillyBlaze/OctoPrint-TouchUI #254: TouchUI Install helper - RC feedback

BillyBlaze, 2018-11-11:
Hi TouchUI-ers,

I have been busy with a script that will automate the process of installing TouchUI locally, I am now ready to release this but before that happens I want to have some feedback.

Follow the guide:
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser

Please let me know if this works or not! ofcourse all other feedback is welcome 👍

sferris, 2018-11-12:
This worked perfectly for me on a brand new install: octopi-stretch-lite-0.15.1-20180418.img

Thanks @BillyBlaze, Great work!

yet-another-average-joe, 2018-11-15:
Hello,

not sure what feedback you are looking for... TouchUI has always been working great for me. Except I had to downgrade to OctoPrint 1.3.8 for previous TouchUI version, and disable OctoPrint auto update. Last release now works perfect with OctoPrint 1.3.9..

So, Yesterday, I did a fresh install, and followed the steps explained here, as always :

https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser-(OctoPi-or-Jessie-Light)

Xserver, LCD declaration, autostart git clone, etc.

It worked fine, as usual : the RasPi boots to browser/TouchUI. I did this successfully maybe 10 or 15 times for the last months ; I tortured tested it many times : *NO PROBLEMS* (except some glitches with multitouch pinch, certainly not related to TouchUI itself, and and a problem with the screen saver I am still trying to reproduce !)

Then, I followed the steps explained there :

https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser :

These commands return fatal errors :

git checkout maintenance -> not a git repository
sudo ~/TouchUI-autostart/helpers/install -> not found

Until now, I had no time to browse directories ; will have a look later.

BillyBlaze, 2018-11-15:
Thanks for the feedback @yet-another-average-joe 

I forgot to add that you need to be active in the directory `~/TouchUI-autostart/` before running the `git checkout` command

I have updated the guide for now with this extra step! Can you try it again?

BillyBlaze, 2018-11-15:
Also thanks for the feedback @sferris ! 👍

yet-another-average-joe, 2018-11-15:
Ok ! At first, I did'nt understand where the "boot to TouchUI" cmd line had to be done, as it did'nt work just after plugin install. So I installed it the classic way, and did'nt understand why it had to be installed 2 time !

Now, it works fine, except one thing : OctoPi does not come with GL driver : error message (not a fatal error). It cannot be enabled with raspi-config, it has to be installed before activation. Maybe you could inform the user ?

I use 2 µSD cards, and play with different backup/ install images. On my printing card, Pixel is installed, and no GL came with. On the testing one, I just installed Pixel, as usually, and GL came with. I was able to activate it. (this being said, I see no difference). (I don't launch TouchUI from Pixel, TouchUI and OctoPrint services are killed before I sudo Pixel). This is weird...

Sorry for my bad english, and thanks for your work !

BillyBlaze, 2018-11-18:
Thanks for the feedback, the installation of the GLDriver is a bugfix for WaveShare devices. I don't want to complicate things for users, so I will make this silent. You can already disable the installation of the GLDriver by using the `-d` or `--disable-gldriver` parameter.

BillyBlaze, 2018-11-18:
I have released this script, thanks again for your feedback!

Dont forget to switch your branch to master!

closed 2018-11-18 by BillyBlaze