BillyBlaze/OctoPrint-TouchUI #255: No Touch and screen timeout

matty0511, 2018-11-12:
I purchased a 7 inch touch screen (http://a.co/d/0EkDweD) my Raspberry PI.  I got the PI to auto boot into a browser but the touch screen is not working. Also, It seems like after a short period of time it goes into a black screen saver mode and being i cant touch the screen, i cant wake it up.

BillyBlaze, 2018-11-12:
1. It sounds like a hardware or driver issue.
   - Check if USB Touch device is recognized (`lsusb`)
   - Check if touch is received (`sudo lsinput | grep -i touch -A4 -B5`)
2. You can disable the screensaver in `/etc/default/touchui` by chaning DISABLE_SCREENSAVER to true

matty0511, 2018-11-14:
okay i found the disable screen saver part but this is waht i get when i run the touch command:
pi@octoprint:~ $ sudo lsinput | grep -i touch -A4 -B5
sudo: lsinput: command not found

BillyBlaze, 2018-11-18:
I forgot you need evtest for this and not lsinput.

- Install it with `sudo apt-get install evtest`
- Run `evtest` and interact with your touchscreen to see if it receives the touch events

closed 2018-12-09 by stale[bot]