BillyBlaze/OctoPrint-TouchUI #256: TouchScreen Calibration won't start

Freakadude, 2018-11-13:
Hi,

Running latest octopie on RPI3 with Raspbian Stretch. Installed the plugin from the repo and have done the touchscreen setup as per Github instructions. All went without any errors. After intial reboot touchscreen shows the touch UI interface only the touchscreen is not responding so i thought i'd do the calibration.

After shutting down the Touch UI service i try to run the calibration tool via sudo xinit ~/TouchUI-autostart/helpers/calibration.xinit.

I get following error: 
Error: No calibratable devices found.
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

Just to check the claim of the error message i started the service again and if i try to run the calibration while TouchUI is running i get error that sais it cannot run since the TouchScreen Display 0 is already in use and that i shall try if its not running. After stopping the service and trying to run calibration again i get the same error with No calibratable devices found.

Freakadude, 2018-11-13:
Nevermind. I realized just now that i forgot to install the LCD drivers as well. Once i had done that the touchscreen was responsive already and i could also run the calibration tool.

BillyBlaze, 2018-11-14:
Awesome! Glad you were able to figure out this problem!

closed 2018-11-14 by BillyBlaze