BillyBlaze/OctoPrint-TouchUI #257: Problems when deactivating/reactivating and deinstalling/reinstalling TouchUI plugin

yet-another-average-joe, 2018-11-16:
Hello,

WARNING : sadistic experiments below

I am currently deactivating/reactivating and deinstalling/reinstalling TouchUI (torture testing).

I first crashed OctoPrint, but unfortunately cannot reproduce the crash. But intresting things happen anyway.

**experiment n° 1 : deactivating / reactivating TouchUI**

1 - connect to OctoPrint+TouchUI from a phone and a computer, TouchUI being toggled ON on the computer and the phone ; 2 instances on the computer : 192.168.xxx.xxx with TouchUI interface and octopi.local with TouchUI. Firefox everywhere, phone and computer.
2 - deactivate TouchUI ; OctoPrint restarts service
3 - refresh the browsers, on the computer and the phone

Everything is fine : phone and computer ; TouchUI interface is gone. TouchUI is still there on the LCD Touch with the dialog box asking to reload : hit reload ! TouchUI attempts to restart, but it can't and displays what I call a RTOD (Red TouchUI of Death) after a while.

The RTOD could be ennoying : it is not possible to take control locally on the Pi with a keyboard. The only solution is to SSH.. Rebooting the Pi does not solve the problem. The TouchUI autostart is not disabled.

4 - reactivate TouchUI in the plugin manager : now you get a BTOD (Blue TouchUI of Death) on the Pi and VNC indeed) : "Loading OctoPrint" forever. Sometime it is a RTOD... But everything is fine on the phone and the computer.

5 - restart the Pi (SSH or browser). Everything if fine on the phone and the computer (after a while)

Conclusion 1 : deactivating TouchUI does not disable X/Chromium/TouchUI autostart, and it fails ; reactivating is not enough : it needs a reboot.

**experiment n° 2 : deinstalling / reinstalling TouchUI**

Initial conditions remains the same.

1 - uninstall TouchUI plugin ; OctoPrint restarts service
2 - refresh the browsers, on the computer and the phone

Everything is fine on the phone and the 2 browser instances on the computer. Meanwhile, on the LCD Touch, and the screen saver is back ! (I'm a screen saver hunter) ; Touch the LCD, then hit reload -> BTOD + screen saver.

3 - reboot system (SSH or browser)

Everything is fine : phone and computer ; RTOD on the LCD touch

4 - reinstall TouchUI ; OctoPrint restarts service

Everything is fine on the 2 browser instances on the computer : TouchUI is back. RTOD is still there on the LCD Touch. On the phone, you get gigantic interface. Huge buttons, and cannot pinch to reduce. Refreshing does not solve the problem. But closing / reopening Firefox mobile does.

5 - reboot the Pi

Everything is fine on the 2 browser instances on the computer : TouchUI is there. LCD touch OK, but once again the gigantic interface on the phone. Huge buttons, and cannot pinch to reduce. Refreshing does not solve the problem. But closing / reopening Firefox mobile does. And the screen saver is gone.

Same conclusion : when deinstalling TouchUI, the autostart should be disabled. I had a look to services, and TouchUI service seems not to be running after deactivation or deinstallation. Deinstalling does not 100% deinstall, and deactivation does not 100% deactivate.

Hope this could help...

(this being said, who will deactivate or uninstall TouchUI in the real life, and have to connect a keyboard to the Pi ?)

BillyBlaze, 2018-11-18:
Hello,

TouchUI and the bootloader are two separate apps that cannot be installed in one go, therefor uninstalling TouchUI will also not disable the bootloader. The first reason is because people will install TouchUI also for other use cases, secondly handling `sudo` within OctoPrint is a hassle, since it's sand-boxed and running as a user, and not on root level.

Like you said you can disable the bootloader by changing the parameter but you could also disable it by removing it from update-rc.d with `sudo update-rc.d touchui remove`.

I wont fix anything for this; simply because if you have an app installed on your phone that communicates with a server (e.g. Plex) then if you remove the server, the client will still be manually removed.

However I will add a script to the bootloader that allows an user to easily uninstall it.

yet-another-average-joe, 2018-11-18:
Thnks for the answer !

closed 2018-11-18 by BillyBlaze