BillyBlaze/OctoPrint-TouchUI #259: New installer : *SEEMS* to be hanging

yet-another-average-joe, 2018-11-19:
Hello BiilyBlaze.

this new install procedure is a huge improvement, but some information is missing on the Wiki page.

After a new OctoPi image is flashed, the Pi is booted. I did this 5 times.

"1 - Make sure you've installed OctoPrint and installed TouchUI from the 'Plugins Manager'".

do you mean the config.txt stuff ? Cannot test with official LCD, mine is HDMI.

"2 - Make sure you've your touchscreen working by installing the required drivers."

"3 - Get the TouchUI boot files + install scripts:" OK

4 - Run the install helper

The plugin is installed from OctoPrint and the installer SEEMS to hang at : 

Detected Stretch Build
Enabling GLDriver 'G1 GL (full MKS)' for Stretch :

Hitting a key solves the problem. But powering off (after 15mn) made the SD card unreadable by Win10. (had to format with my DSLR)

Maybe an error message (no GL), asking for hitting a key, and hidden by the silent install ? (with the OctoPi image, GL is not installed)

Using the -d switch also solves the problem, but you forgot it in the Wiki, and closed the thread where you gave the information...

Could the installer detect if GL is there or not ?

5 - done, and pi99-fbdev.conf has been created, with the parameters for my HDMI LCD touch 
Everything"s fine !

A few qustions : 

- does the installer detect the display (Option "fbdev" in 99-fbdev.conf, 0 or 1)
- is there a switch for verbose installation ?

BillyBlaze, 2018-11-19:
Thanks for creating the issue.

You can disable silent mode by adding `--debug` parameter, that will show the whole output. I will add some documentation, but all parameters are also explained under `--help`.

I will have a look into this GL issue. I think this is causing you these issues.

yet-another-average-joe, 2018-11-19:
OK, thanks for the informations.

I use a 7" 52Pi LCD capacitive touch. It is correctly recognized !

yet-another-average-joe, 2018-11-22:
Here is the hidden error message, if -d or --disable-gldriver switch is not set :

[image]

closed 2019-05-15 by BillyBlaze