BillyBlaze/OctoPrint-TouchUI #260: Virtual keyboard and Enter key

yet-another-average-joe, 2018-11-19:
Hello,

Enter key is not working with the virtual keyboard. It should the way the green key works...

(Esc does work)

Tested with a keyboard on the Pi, and with VNC Viewer. (VNC is great with TouchUI)

BillyBlaze, 2018-11-19:
Hi,

If you have a keyboard attached then I suggest to disable the virtual keyboard in the "TouchUI Settings"

yet-another-average-joe, 2018-11-19:
Why ? It works fine, except for the Enter Key ! It does'nt hurt !

This being said, VNC is great. The TouchUI GUI is far better than the default one. I connected a keyboard to the Pi just for verification purpose, because Enter key did'nt work with VNC and I wanted to know if it was a VNC issue or a TouchUI one. Using the LCD Touch and VNC meens that the virtual keyboard has to be enabled. Switching it on and off is tedious.

Everything can be done with TouchUI. Except one. The only missing feature is drag/drop for GCode files (VNC cannot do this, and for TouchUI in the browser : I don't know). The workaround is Samba and the watched folder.

Imagine this use :

- an LCD even without touch feature, and TouchUI
- a BT mini keyboard with pointing device
- TouchUI with VNC on computers

This is universal. Just the Enter key problem !

TouchUI + VNC is the way to go on a computer!

[vnc](https://user-images.githubusercontent.com/43514870/48725183-a7eee400-ec2b-11e8-85fd-598909cedbc1.PNG)
.

closed 2018-12-10 by stale[bot]