BillyBlaze/OctoPrint-TouchUI #265: I have a windows with login just before to enter in Touch Ui

thsrp, 2018-12-11:
Each time I power on the Pi connected to the official screen 7" Raspberry, just before to enter in the Touch Ui I have a windows with login and password. Im obliged to connect a keyboard to my Pi to enter the login and the password ... I ticked the box "remember" but it is useless. Is there no way to stop having this login window and have directly on the screen Raspberry the interface "Touch Ui"...?
Attention, I am a "noob" so explain me with simple words ..! Thank you

bolsoncerrado, 2018-12-12:
It's because of the FORCE LOGIN plugin. 

I'm also interested in a solution to this and NOT just disable this security feature.

thsrp, 2018-12-12:
Force login Plug in ??? where can I find this plug in? 
It seems very compelling to have permanently a keyboard plugged into the "Raspberry" just to have access to the interface "touch ui" on the screen connected to "raspberry" ?? no ?

bolsoncerrado, 2018-12-12:
In the Plugins Manager under Settings in your Octoprint instance.

Yes, a virtual keyboard should be allowed to type any login/pass on this much needed login screen (security feature!)

bolsoncerrado, 2018-12-12:
BTW its unrelated to TouchUI but to the Force Plugin instead:
https://github.com/OctoPrint/OctoPrint-ForceLogin/issues/1

BillyBlaze, 2018-12-12:
I have looked into to attach the TouchUI virtual keyboard in ForceLogin 'window', but there is currently no method to inject it.

For your raspberry pi, I suggest to use the autoLoginAs feature in OctoPrint, if you install TouchUI with the installation helper, it will ask you for a username to set this automatically up.

You can add this manually by editing `~/.octoprint/config.yaml`:
1. `nano ~/.octoprint/config.yaml`
1. `add the following under accessControl:`:
  - ```
       autologinLocal: true
       autologinAs: [YOUR-USERNAME]
     ```
1. Replace `[YOUR-USERNAME]` with the user you want to auto login with
1. Save and Exit (CTR-O, CTR-X)

thsrp, 2018-12-12:
Hi, many thanks Billy , I'm going to edit manually ... Is it possible to edit from my computer, with SSH (Putty) ?

bolsoncerrado, 2018-12-12:
I did use the helper with our login but it didn't work either.

BillyBlaze, 2018-12-12:
@thsrp 
yes

@bolsoncerrado 
If you have any issues with the autoLoginAs feature, then you can best report in the OctoPrint repository, since it's core functionality. However before doing that, does your octoprint log contain any errors?

bolsoncerrado, 2018-12-12:
Seems I may have forgotten to restart the service. Now working! :P

BillyBlaze, 2018-12-12:
That sounds like the issue indeed :) 

Did you select 'no' when the install helper asked to reboot OctoPrint? Or didn't it ask that at all?

bolsoncerrado, 2018-12-12:
It did ask to, but now that you mention it, it was too fast, aka i found the touch screen resting on the command line "login" screen, aka I think it actually never rebooted. Are you invoking the reboot with sudo?

BillyBlaze, 2018-12-12:
1. The installer can only run as root.
2. It will not reboot, it will only restart the OctoPrint service (when setting autoLoginAs) and start TouchUI service when the installer successfully installed.

bolsoncerrado, 2018-12-12:
Then something's wrong or why am I kicked to the terminal login screen (text mode) instead of TouchUI?

BillyBlaze, 2018-12-12:
Maybe the installer was unsuccessful in the last step and didn't launch TouchUI until you rebooted.

BillyBlaze, 2018-12-12:
make sure you have two white-space characters before autologinLocal and autologinAs and that you replace YOUR-USERNAME with a valid user that can login into OctoPrint.

make92, 2018-12-15:
hi, same problem for me. getting a login screen for TouchUI even having the below in the 
 /home/pi/.octoprint/config.yaml

accessControl:
  autologinLocal: true
  autologinAs: octoprint

bolsoncerrado, 2018-12-15:
Well Im now having the weirdest issue ever....

I can access Octoprint in desktop remote mode in any computer on the LAN or outside of it, launch prints, monitor etc. and I can access Octoprint in local touch screen **BUT** IT WON'T REPORT WHAT THE PRINTER IS DOING AT ALL!! :???? However I can change temps and switch tabs! I don't get it!??! Only in LOCAL is not working :????

thsrp, 2018-12-15:
>           hi, same problem for me. getting a login screen for TouchUI even having the below in the
> /home/pi/.octoprint/config.yaml
> accessControl:
> autologinLocal: true
> autologinAs: octoprint

as BillyBlaze say : 2 white-space characters before the two lines autologinlocal….. and autologinAs

make92, 2018-12-15:
> > ```
> >       hi, same problem for me. getting a login screen for TouchUI even having the below in the
> > ```
> > /home/pi/.octoprint/config.yaml
> > accessControl:
> > autologinLocal: true
> > autologinAs: octoprint
> 
> as BillyBlaze say : 2 white-space characters before the two lines autologinlocal….. and autologinAs

sorry was a copy&past mistake, checked the file and i have the 2 white-spaces in front of both lines. but still the same problem :(

BillyBlaze, 2018-12-16:
@make92 
Are you connecting with IPv6? (if you see TouchUI loading it should say in small letters in the bottom of the screen 'Connecting to ::1' or 'Connecting to localhost')

bolsoncerrado, 2018-12-19:
You're probably talking of the force plugin that comes embeded with Octoprint 1.3.10 now....

BillyBlaze, 2018-12-19:
Yes, he also connects with IPv6, which seems exactly the same issue as #266

bougs93, 2018-12-19:
Thank you.
I connected to USB keyboard and logged in and rebooted.
Success.

S-Bartfast, 2019-01-04:
I am having the same problem as make92 (auto login not working).

> Are you connecting with IPv6? (if you see TouchUI loading it should say in small letters in the bottom of the screen 'Connecting to ::1' or 'Connecting to localhost')

Mine shows the IPv6 connection ("::1")

S-Bartfast, 2019-01-05:
Hmm, seems to be working now. No idea why it wasn't before.

closed 2019-01-30 by stale[bot]

jglazer63, 2019-03-03:
Me too.  Can someone please lay out the steps to correct this?

darkcurrent, 2019-03-04:
I am using 3.5 TFT touch screen with TouchUI. I want to remove log in screen from TouchUI start. I have tested several config.yaml configuration but none of them is working. The "log-in screen" appears all the time.

```
accessControl:
  autologinLocal: true
  autologinAs: myUserName
  salt: mySaltCode

accessControl:
  enabled: false
  firstRun: false
  autologinLocal: true
  autologinAs: myUserName
  salt: mySaltCode

accessControl:
  enabled: false
  firstRun: false
  #autologinLocal: true
  #autologinAs: myUserName
  #salt: mySaltCode
```

Rhynier, 2019-03-18:
Adding the 'localNetworks' section with both the IPv4 and IPv6 localhost values worked for me. Before it was also navigating to the login page.