BillyBlaze/OctoPrint-TouchUI #268: Custom tab icons for other plugins

LazaroFilm, 2018-12-18:
Hi,
Love touch UI on my tablet but I have a few plugins that generate extra tabs and they all show up as a big question mark. I'd love a way to turn those ? into an custom icon. Maybe make it selectable from the [fowntawesome.com](https://fontawesome.com/cheatsheet?from=io) list.
Thanks!

BillyBlaze, 2018-12-18:
Thanks for reporting, the intention of the question mark was to be clear that the plugin is not officially supported in TouchUI, however I'll revisited this and try another solution in the next version.

LazaroFilm, 2019-02-12:
An idea would be to allow to use the fontawesome icons instead of the ?. 
https://fontawesome.com/cheatsheet?from=io

LazaroFilm, 2019-02-19:
Hi, any update on this? One thing I was thinking is, Im also using a plugin called Tab Order which allows to assign icons to each tabs. If there was a way for you to have these icons used instead of the default that would be an easy fix.

BillyBlaze, 2019-05-15:
Hi @LazaroFilm,

Sorry for the long delay, I have tried to implement a more automatic system, but after a few hours I concluded that this would require to much time, so I think your last suggestion is the best way forward :) Perhaps we could start with the list you have in the upcoming release.

closed 2019-05-25 by BillyBlaze

cappy0815, 2020-01-17:
Hi,
I checked it but the icons chosen with TabOrder do not appear at TouchUI :-(

MrCrin, 2020-02-20:
Can I add my voice to the mix? Is there a way to add these icons manually ourselves when one pops up with no icon?

I currently have File Manager, Octoplapse and Slicer prodding at my eyes with "? ? ?"

buzkill420, 2020-03-01:
Dear Billy! I am having the issue where anytime i restart TouchUI or OctoPrint, the Tab Navigation Bar resizes much longer then the screen and i am unable to select the Settings tab, as its defaulted to the far right....is there any way to shrink these or to force the scaling? thanks

commit 133ac6e6 (2019-05-25): :art: Add plugin icons

Add some random icons for plugins

Fixes #268