BillyBlaze/OctoPrint-TouchUI #272: Unable to Login at all using touchUI on 3.5" screen PI B

SandwichRising, 2018-12-29:
OctoPrint 1.3.10
TouchUI 0.3.12
octopi 0.15.1

I have competed setup of touchui on my pi, and am able to login using my credentials using a PC over the network. I have also setup the touchui to auto-load on my pi on startup. When it finishes loading it prompts me to "Please log in" even though I set it to autologin my user. My Octopi login doesn't work, it gives me an "Incorrect name or password" error, and does the same thing with my pi su login as well. However I am using the same credentials I use to login with a pc over the network and it appears to be logging in passively just fine. 

I have attempted editing my ~/.octoprint/config.yaml to the following:
AccessControl:
  autologinAs: myoctopilogin
  autologinLocal: true
  localNetworks:
  - 127.0.0.0/8
  - ::1/128
  salt: 4zwrHYQ3BoWqDbSWYIyHIGhM29sn7jRW
api:
  key: 

But when I try to login a couple times and check the file again is has been updated to:
AccessControl:
  autologinAs: myoctopilogin
  autologinLocal: true
  localNetworks:
  - 127.0.0.0/8
  - ::1/128
  salt: 4zwrHYQ3BoWqDbSWYIyHIGhM29sn7jRW
accessControl:
  salt: EuN40TnP6BbPn1hF2mwTtH5coHxDOdMP
api:
  key: 75D081F106BA4B2AA400140CB5D2DBEA

So it adds the extra accessControl: line but only after I try to login, and I'm not sure why. Also, that's not my actual user api key from inside octoprint, and adding it from the used settings to that file doesn't seem to help. I am restarting both octoprint and touchui after any changes are made.

In the log it looks like I'm getting some sort of 401 POST error. My octoprint.log is [here](https://pastebin.com/GQmnUnd3) and also attached.

[octoprint.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/2716762/octoprint.log)

BillyBlaze, 2018-12-30:
Make sure you have the right whitespaces (2 before autoLoginAs etc, and 4 underneath `localnetworks:`. Also it's `accessControl` and not `AccessControl`

it should look something like this:

```
accessControl:
  autologinAs: myoctopilogin
  autologinLocal: true
  salt: somelonghash
  localnetworks:
    - "127.0.0.1/8"
    - "::1/128"
```

SandwichRising, 2018-12-30:
Yeah, githubs autoformat was getting rid of my spaces. And I think because mine has "Access" instead of "access" that's probably the root of all my problems, and I'm not sure how it happened. And that's why it keeps remaking it when I try to log in. I will make the changes and try again, thanks! Ill keep you updated.

SandwichRising, 2018-12-31:
Okay, that was the root of the problem. Now I've got it auto logging in. My touchscreen stopped working at that point, so now I have to figure out if that's hardware or software. Thank you for the help though, my original issue is resolved.

SandwichRising, 2019-01-01:
I damaged my touchscreen hardware, but have it replaced now. All my issues are solved. Thank you for the help, BillyBlaze!

jeor112, 2019-01-07:
Hi BillyBlaze
I have the same issue. I have tried to reinstalling a few times with the same results.
I have tried to do follow this steps and inserted this:
  localnetworks:
    - "127.0.0.1/8"
    - "::1/128"
in the config.yaml file, but there is still no autologin.

Now, the config file looks like this:

accessControl:
  autologinLocal: true
  autologinAs: " myusername "
  salt: " somelonghash "
  localnetworks:
    - "127.0.0.1/8"
    - "::1/128"
api:
  key: " It is the correct API key "
plugins:
  announcements:
    _config_version: 1
    channels:
      _blog:
        read_until: 1545132000
      _important:
        read_until: 1521111600
      _octopi:
        read_until: 1527588900

To be able to login, I will have to connect USB keyboard :-(

I hope you can help!
Thx in advance

docb7, 2019-01-12:
Hi, I have the same issue. I did not have Themeify installed but I took the chance and deactivated all Plugins I had, but nothing helps.  My config.yaml looks good and the autologin works from a normal browser. But not via Touch UI on the octopi. Also there is no virtual keyboard and when I attach a real keyboard I always get username/password wrong. Anyone got an idea what went wrong?
Ocotpi is 1.3.10, TouchUI 0.3.12

SandwichRising, 2019-01-13:
One of the problems I had with the pi was passwords. My pi defaults to a UK layout typically, and I have to tell it to use a US layout instead. The symbols are at different locations for both layouts which was the root of my password problems.

SandwichRising, 2019-01-14:
Did you add the api key from octoprint?

docb7, 2019-01-14:
Nope, where can I do that? I have installed touch ui on the same raspi as Octopi runs on, is there still a need for putting the api key somewhere? If yes - where?

docb7, 2019-01-14:
Do I need to put the Octoprint Api Key into the  config.yaml?

SandwichRising, 2019-01-15:
I put the OctoPrint api from inside the OctoPrint interface into the config.yaml on mine before it would autologin.

docb7, 2019-01-16:
Ah, that was done automatically somehow. The api key is in config.yaml so that is not the error.

sbias, 2019-01-17:
try to set AUTODETECT_PORT=false in /etc/default/touchui . Don't know why (yet) ... but this solved the problem for me.

docb7, 2019-01-18:
Real crazy, after about 10 restarts (cause I shutdown the system when not printing) the login works. I have changed nothing ;-) Really funny. Thanks for all your support!

jglazer63, 2019-02-09:
Still having this issue.   Where is the api?

closed 2019-03-02 by stale[bot]