BillyBlaze/OctoPrint-TouchUI #274: [Plugin Conflict] OctoPrint-ConolidateTempControl

jneilliii, 2019-01-01:
Trying to figure out a way to get around this problem and have yet been able to come up with anything @BillyBlaze. I have an unpublished plugin that a handful of people are using and have come across a conflict with TouchUI that ends up totally garbling the UI.  I have created a branch that has my current efforts of not processing when TouchUI is loaded and active at the link below.  Any assistance would be greatly appreciated. 

Is there a simple way to not even load/remove my plugins viewmodel when touchui is enabled, something like [this](https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/source/js/plugins/webcamTab.js) in your source but would run within my plugin code?

BillyBlaze, 2019-01-16:
hi @jneilliii 

Sorry for the long delay, I can add your plugin to the remove list in the next release. It already have some important workarounds. e.g. TouchUI will detect viewmodels of plugins before they're activated by OctoPrint but after all plugins register themself. You could ofcourse try to see if TouchUI is active and not register your plugin to OctoPrint, but I would prefer the first option.

jneilliii, 2019-01-16:
I think I figured out an approach on one of my other plugins that seems like it might work. I'll try on the iget plugin this weekend and report back. Something like this...

https://github.com/jneilliii/OctoPrint-StatefulSidebar/blob/master/statefulsidebar/static/js/OctoPrint-StatefulSidebar.js

closed 2019-02-11 by stale[bot]