BillyBlaze/OctoPrint-TouchUI #277: Possible logging issue causing system slow down

leitenky, 2019-01-09:
I reported this a while back but the issue seems to have come back up again.
Out of the blue I had very slow printerhead movement and octonprint seemed to slow way down.  The only way I could resolve it was to uninstall though UI.  I have a log file if you would like but  this is what the 5mb log file looks like:
"2019-01-08 07:00:12,865 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
2019-01-08 07:00:13,577 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
2019-01-08 07:00:14,088 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
2019-01-08 07:00:14,759 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current"

It's just thousands of lines of this

Any support would be greatly appreciated!

adilinden, 2019-01-22:
Just a me too comment. Trying to diagnose some webcam plugin issue, I have a 1.4MB log file primary repeating this over and over:

```
Line 127: 2019-01-22 06:16:30,962 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
Line 128: 2019-01-22 06:16:31,470 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
Line 129: 2019-01-22 06:16:31,978 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
Line 123: 2019-01-22 06:16:32,483 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message 
```

```
Line 7617: 2019-01-22 07:35:26,113 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
Line 7618: 2019-01-22 07:35:26,623 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
Line 7619: 2019-01-22 07:35:27,130 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: current
```

Adi

BillyBlaze, 2019-01-22:
Thanks for reporting, can you tell me a bit more about how TouchUI is running (I presume on a touchscreen attached to the Pi, that is already autoLogedIn as a specific user?)

mateusaguiar, 2019-02-11:
had the same issue here!
the last lines of the log file looks like this:
[image]

leitenky, 2019-02-12:
Hi @BillyBlaze 
sorry for the delay in getting back to you. I am using an official Pi 7" touch screen.
My setup is OctoPi image and I followed your tutorial to set up autoboot. 
After several remove and reinstall of TouchUI coupled with enabling and disabling logging for TouchUI the issue seems to be resolved. Not sure how the logging keeps getting enabled

BillyBlaze, 2019-03-16:
Thanks all again for reporting. I am closing this issue because there are a few duplicates issues and I will continue the investigation in #281

closed 2019-03-16 by BillyBlaze