BillyBlaze/OctoPrint-TouchUI #279: Ran Boot to browser script twice now Pi wont start at all

Warsenius, 2019-01-16:
So i ran the boot to browser script, rebooted the device (triggered by wizzard) but i still got the cli login prompt on the screen.
So i ran the script again, it again said 'TouchUI is successfully installed' and it said:
```
Skipping installation of dependencies:
 - All dependencies are already installed

Skipping creation of xserver config file:
 - /dev/fb1 already configured

Move touchui.init to '/etc/init.d/touchui':
 - success

Move touchui.default to '/etc/default/touchui':
 - success

Register touchui:
 - success

Skipping setup autoLogin for localhost:
 - Login as 'Warsenius' already configured

TouchUI is successfully installed, Do not remove the directory
~/TouchUI-autostart and if you change the path of this directory
then edit the file /etc/default/touchui with the new path.

Detected Stretch build!
Enabling GLDriver 'G1 GL (Full KMS)' for Stretch:

Launching TouchUI:
 - success
```

i rebooted the pi again but now it wont start up at all.
its not pingable, and nothing shows up on hdmi or touchscreen.
any ideas on how to troubleshoot this?

BillyBlaze, 2019-01-16:
Hi @Warsenius,

Can you share the contents of the file `/var/log/xorg.0.log`, that should tell us a bit more

Warsenius, 2019-01-16:
i will see if i can get access to the sdcard in any way to get the data out

Warsenius, 2019-01-17:
```
[   367.025] 
X.Org X Server 1.19.2
Release Date: 2017-03-02
[   367.026] X Protocol Version 11, Revision 0
[   367.026] Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
[   367.026] Current Operating System: Linux octopi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l
[   367.026] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=25ab0ce2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait fbcon=map:10 fbcon=font:ProFont6x11
[   367.026] Build Date: 18 October 2017  04:55:30PM
[   367.026] xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support) 
[   367.026] Current version of pixman: 0.34.0
[   367.026] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   367.026] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   367.027] (==) Log file: "/var/log/Xorg.0.log", Time: Wed Jan 16 17:11:24 2019
[   367.029] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   367.029] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   367.030] Parse error on line 10 of section InputClass in file /etc/X11/xorg.conf.d/99-calibration.conf
	The Section keyword requires a quoted string to follow it.
[   367.030] (EE) Problem parsing the config file
[   367.030] (EE) Error parsing the config file
[   367.030] (EE) 
Fatal server error:
[   367.030] (EE) no screens found(EE) 
[   367.030] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   367.030] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   367.030] (EE) 
[   367.031] (EE) Server terminated with error (1). Closing log file.
```

BillyBlaze, 2019-01-17:
```
Parse error on line 10 of section InputClass in file /etc/X11/xorg.conf.d/99-calibration.conf
```

Did you manually edit this file or copy-pasted code into that file? Typically this error is thrown if the wrong double-quotation is used or that the config is missing an `EndSection`

Warsenius, 2019-01-17:
Ye i did manually edit it. was following this guide and seems like the quotes didnt copy right
http://www.plastibots.com/index.php/2018/06/02/adding-tft-lcd-to-raspberry-pi-octopi-rig-touchui/

but this would not explain why the Pi wont boot up from this sdcard anymore after running the script twice.

BillyBlaze, 2019-01-17:
Yes, that guide is notorious for not being able to use the right quotation marks. It's pretty annoying. (especially when I asked the author that he could let TouchUI do this, but I am not getting any response)

I also have no idea why your system is not starting anymore, TouchUI do not do any manipulation on system level and I am able to run the installer over and over again.

We could debug this, but maybe a fresh re-install might be a better and faster idea?

Warsenius, 2019-01-17:
yea i will have to buy a new sdcard tomorrow so i can fresh install on that. and then hopefully recover my settings and plugin files

Warsenius, 2019-01-19:
so flashing the new sdcard didnt work. bought a new Pi and it booted up with the old sdcard.
but now i'm hesitant to put the touchscreen on it dont want to break it.

closed 2019-02-11 by stale[bot]