BillyBlaze/OctoPrint-TouchUI #280: TouchUI opens always, regardless if desktop or mobile

pandel, 2019-01-18:
Hi!

I recently added TouchUI to my fresh OctoPi installation. But sadly, it open always, regardless if I open Octoprint via desktop browser (Chrome or Firefox) or via mobile phone.

My currently installed modules are:
```
argh (0.26.2)
awesome-slugify (1.6.5)
Babel (2.5.3)
backports-abc (0.5)
backports.ssl-match-hostname (3.5.0.1)
Bed-Visualizer (0.1.7)
blinker (1.4)
certifi (2018.4.16)
chainmap (1.0.2)
chardet (3.0.4)
click (6.2)
emoji (0.4.5)
feedparser (5.2.1)
Flask (0.10.1)
Flask-Assets (0.10)
Flask-Babel (0.9)
Flask-Login (0.2.11)
Flask-Principal (0.3.5)
frozendict (1.2)
future (0.15.2)
futures (3.1.1)
idna (2.6)
itsdangerous (0.24)
Jinja2 (2.8.1)
Markdown (2.6.11)
MarkupSafe (1.0)
monotonic (1.3)
Nautilus (1.22)
netaddr (0.7.19)
netifaces (0.10.7)
numpy (1.16.0)
OctoPrint (1.3.10)
OctoPrint-Cancelobject (0.2.1)
OctoPrint-DragonOrder (0.1.2)
OctoPrint-EEprom-Marlin (1.2.1)
OctoPrint-Estop (0.1.2)
OctoPrint-Fullscreen (0.0.4)
OctoPrint-PortLister (0.1.8)
OctoPrint-SystemCommandEditor (0.3.2)
OctoPrint-Tasmota (0.8.5)
OctoPrint-Telegram (1.4.2)
pathtools (0.1.2)
Pillow (5.4.1)
pip (9.0.3)
pkg-resources (0.0.0)
pkginfo (1.4.2)
psutil (5.4.8)
pyasn1 (0.4.2)
pybonjour (1.1.1)
pylru (1.0.9)
pyserial (3.4)
python-dateutil (2.6.1)
pytz (2018.4)
PyYAML (3.10)
regex (2018.2.21)
requests (2.18.4)
rsa (3.2.3)
sarge (0.1.5.post0)
scandir (1.3)
semantic-version (2.4.2)
setuptools (39.1.0)
singledispatch (3.4.0.3)
six (1.11.0)
sockjs-tornado (1.0.3)
speaklater (1.3)
Stateful-Sidebar (0.1.1)
Themeify (1.2.0)
tornado (4.5.3)
TouchUI (0.3.12)
Unidecode (0.4.21)
urllib3 (1.22)
watchdog (0.8.3)
webassets (0.12.1)
websocket-client (0.40.0)
Werkzeug (0.8.3)
wheel (0.31.0)
wrapt (1.10.11)
```

I just installed OctoPi 0.15.1 and added the different plugins via Pluginmanager. OS itself is up-to-date via apt-get update/upgrade.

What can I do? I have to manually en-/disable TouchUI and this is quite annoying ;-)

Regards,
Holger

ewortzman, 2019-01-20:
Having the same issue. Only way to resolve it is to manually disable TouchUI

BillyBlaze, 2019-01-24:
Thanks for reporting.

There are two conditions on when TouchUI activates, both of these conditions can be disabled. The first condition is if the device register itself as "touch" capable device, although this can be a false positive. The second condition is resolution; anything below 980 will active TouchUI.

If both conditions are disabled then you can still activate it by adding `?touch` in the url

You can disable these conditions individually under 'Settings' > 'TouchUI'.

@pandel 
If you disable this conditions, do still have this issue?

closed 2019-01-24 by pandel