BillyBlaze/OctoPrint-TouchUI #281: TouchUI stop refreshing HDMI Display

maganzi, 2019-01-20:
Hey,

i have found the following issue.
I can start prints at my Macbook, everything is working fine.
I can control the printer with the OctoPrint itself on the HDMI Touchscreen. But i can´t start a new print. The HDMI Display isn´t frozen but it doesn´t refresh the printing item area.

Thank you for your help!

[octoprint.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/2777060/octoprint.log)

[image]

maganzi, 2019-02-08:
still have this problem

Jaxel, 2019-02-25:
I have the same issue.

BillyBlaze, 2019-03-11:
Thanks all for reporting this issue. I am trying to reproduce the issue but sofar I have no success. Could anyone please share your octoprint.log so I could possible identify a suspect.

TheNore, 2019-03-14:
[octoprint.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/2968050/octoprint.log)

BillyBlaze, 2019-03-14:
hmm, seems like everyone who has this issue have a big log file with a repeating `Socket message held back, but backlog full. Throwing message away`.

Could you confirm how you login? Do you have autoLoginAs enabled or do you manually login with TouchUI?

TheNore, 2019-03-14:
I have logged in with touch UI when prompted, then saved PW.

BillyBlaze, 2019-03-16:
Hi @foosel, sorry to intrude you with this issue on this tracker, but I am unable to find a good way to reproduce the following issue and to find out if this is a OctoPrint issue:

1. It seems the ForceLogin plugin is blocking the WebSockets while users are logged in and past the ForceLogin screen
2. I have looked into foosel/OctoPrint/issues/2898 but seems not related

Do you want to move this issue over to the OctoPrint issue tracker to have more information?

foosel, 2019-03-17:
> Socket message held back, but backlog full. Throwing message away

So that indicates that the web frontend did not authenticate with the websocket. It should do that immediately after (passive) login. For some reason it is not doing it, which makes me think it might be an issue of some stale version of the web interface in the browser's cache that doesn't yet do the authentication. I've had two people run into this now due to a weird situation in their OctoPrint install, caused by some old files lying around, see [this FAQ entry](https://community.octoprint.org/t/the-web-interface-seems-stuck-not-responsive-broken/5675). Maybe this is a similar situation?

> It seems the ForceLogin plugin is blocking the WebSockets while users are logged in and past the ForceLogin screen

This is true. The fix for foosel/OctoPrint#2898 introduced the message backlog in order to make this process less side effect heavy for third party plugins. This backlog usually should not fill up - that it does indicates that for some reason the socket authentication is not happening. Does TouchUI rely on any kind of API key based authentication or does it just use stock auth?

BillyBlaze, 2019-03-17:
Thanks for your time and the response!

> I've had two people run into this now due to a weird situation in their OctoPrint install, caused by some old files lying around, see this FAQ entry. Maybe this is a similar situation?

This could very well be the issue, there have been some issues with caching before and a latest [Pull Request](https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/commit/a642b33c25d29c8535ee6210d8ace56e1d965602) disables the entire cache.

> Does TouchUI rely on any kind of API key based authentication or does it just use stock auth?

No, TouchUI does not rely on API keys and use stock auth. The philosophy of TouchUI is to transfer the layout and to minimize any side-effects on OctoPrint core functionality.

BillyBlaze, 2019-03-17:
@all, can you try [updating the bootloader](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader) to the latest version to see if this issue still persists?

Jaxel, 2019-03-17:
I updated the bootloader... but now I'm getting a login screen and I don't know why.

BillyBlaze, 2019-03-17:
Good, this is expected. We only need to setup the autoLogin feature on your touchscreen:

1. Run `sudo ~/TouchUI-autostart/helpers/install`
2. Insert username during autoLogin step

Jaxel, 2019-03-17:
Okay, all is working now. Lets see if the screen dies again in a few days.

TheNore, 2019-03-18:
I will let you know as well. Just updated the bootloader.

Jaxel, 2019-03-19:
It froze again. Worse than before. I used to be able to still navigate the menus, it just never updated the print status. Now I can't navigate the menus at all. The mouse cursor shows up, but nothing changes.

BillyBlaze, 2019-03-19:
That sounds like a different issue; can you start by sharing the latest octoprint.log?

TheNore, 2019-03-24:
this is still happening ;/ little to no improvement.

BillyBlaze, 2019-03-24:
can you share the latest octoprint.log?

TheNore, 2019-03-25:
I have printed a little since. here are the last two. 
[octoprint.log.2019-03-22.txt](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/3003845/octoprint.log.2019-03-22.txt)
[octoprint (2).log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/3003846/octoprint.2.log)

TheNore, 2019-04-08:
this is still happening, - latest log attached.
[octoprint (3).log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/3055273/octoprint.3.log)

BillyBlaze, 2019-04-08:
Could you both disable the plugin 'Cancel Objects'? I see alot of errors with this plugin in your log files.

TheNore, 2019-04-09:
Tried this, will report back

TheNore, 2019-04-11:
Still happening.
[octoprint (4).log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/3066671/octoprint.4.log)

TheNore, 2019-04-25:
this is still happening, although not as much since the plugin "object cancel" disabled.

TheLion, 2019-05-12:
I have the same issue, only not using a HDMI display, but connected to SPI.

When I refresh TouchUI from the settings, all is well again.

TheNore, 2019-05-12:
Just to update. I upgraded my octopi setup to the spi based 7" touch
screen, and this problem is still happening.

TheLion, 2019-05-15:
So it's not really depending on what type or size of screen you connect.

TheNore, 2019-05-15:
This is my experience. I have 2 printers with 7" and one with 3.5". Both
have the same issue.

BillyBlaze, 2019-05-15:
Does this issue also persist on OctoPrint 1.3.11? and whats the octoprint log of that?

TheLion, 2019-05-16:
My preliminary finding is that in Octoprint 1.3.11 is does refresh and doesn't freeze up anymore. :-) But have to test some more.

TheLion, 2019-05-18:
Unfortunately it didn't update again during my last print :-( It does respond to touch, like scrolling, but not all commands (like trying to get to the TouchUI submenu to refresh TouchUI) work.

TheNore, 2019-05-18:
I also can not get to the touch ui settings menu now and that I'm running
on 1.3.11, this is happening on both of my printers.

BillyBlaze, 2019-05-18:
The issue about no able to open the TouchUI settings in 1.3.11 is already fixed (#299)

That OctoPrint stops refreshing is still a concern, any change to share your octoprint log again.

TheLion, 2019-05-18:
The TouchUI settings issue isn't there all the time. After restarting the Pi or TouchUI it works just fine for a while. It is something in the menu in general, also not able to enter the Connection menu item.

TheNore, 2019-05-18:
Here is my latest log

BillyBlaze, 2019-05-20:
I dont see any log :(

TheNore, 2019-05-21:
now?

BillyBlaze, 2019-05-21:
nope, no attachments.

TheNore, 2019-05-21:
[octoprint (7).log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/3203765/octoprint.7.log)

TheNore, 2019-08-17:
This is still happening to two different octo print instances on two different screens

Kiron21, 2019-08-20:
Me too, me too :(

Vertigo1206, 2019-12-16:
Still the same issue.
Needs to be reloaded at least once a day.

Raspberry Pi 4B
Octoprint: Version 1.3.12

Octoprint log attached:
Does the Plugin have its own log file?
[octoprint 16.12.2019.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/3967749/octoprint.16.12.2019.log)

AfroPsycho, 2019-12-23:
Same issue here everything is up-to-date and it just happened a minute ago this isn't the first time it happens by the way
Maybe little refresh button will do the work whenever it happens I restart the octoprint but refreshing the touchui would be easier

lciscon, 2020-01-04:
I am having this issue as well.  Same error message.  

This happens to me usually after the session has been running for some length of time (e.g. hours - but it varies).  Given that, it seems logical to me that the browser session is timing out and there is a race condition occurring when everything tries to reconnect. 

Rather than trying to fix the race condition, perhaps it would be easier to try to either remove or postpone the timeout itself?  Is there a browser setting for that?  Or maybe something could generate ping events simulating user input periodically as a kind of keep-alive?

Just a thought...

TheNore, 2020-01-04:
@BillyBlaze - is there any way we can help? I know this is a old issue that still plagues us. I will gladly donate for your time. Over the past year it has been very tough to use my printer accurately, as I highly depend on the touchui for my printers.

Vertigo1206, 2020-01-05:
I found a lot of arguments, maybe this will held
[https://peter.sh/experiments/chromium-command-line-switches/]([https://peter.sh/experiments/chromium-command-line-switches/])

TheNore, 2020-03-04:
Just upgraded to Octoprint 1.40 and get this screen on the initial touchui startup.
[image]

AfroPsycho, 2020-03-04:
> Same issue here everything is up-to-date and it just happened a minute ago this isn't the first time it happens by the way
> Maybe little refresh button will do the work whenever it happens I restart the octoprint but refreshing the touchui would be easier

I dont know how did overlook this for a long time but in the TouchUI settings there is a "Refresh TouchUI" button which solves my problem completely 
All I need was a little refresh button now when touchUI show outdated information I can simply refresh it

TheNore, 2020-03-04:
ive been doing this for at least 6 months on 2x instances

BillyBlaze, 2020-03-07:
> @lciscon
> I am having this issue as well. Same error message.
> 
> This happens to me usually after the session has been running for some length of time (e.g. hours - but it varies). Given that, it seems logical to me that the browser session is timing out and there is a race condition occurring when everything tries to reconnect.
> 
> Rather than trying to fix the race condition, perhaps it would be easier to try to either remove or postpone the timeout itself? Is there a browser setting for that? Or maybe something could generate ping events simulating user input periodically as a kind of keep-alive?
> 
> Just a thought...

Thanks for your feedback, All your suggestions would be something for OctoPrint because that is responsible for updating the information and keeping the connection alive with its websockets. All this is not managed nor manipulated by TouchUI. We only give it a bit of styling.

> @TheNore 
> @BillyBlaze - is there any way we can help? I know this is a old issue that still plagues us. I will gladly donate for your time. Over the past year it has been very tough to use my printer accurately, as I highly depend on the touchui for my printers.

I really want to resolve this issue, however since I am unable to reproduce this issue I cant really help you. In addition to that; TouchUI doesn't not do manipulate the communication (WebSocket) connection. It's entirely managed by OctoPrint so if we do find the issue, I am afraid we need to fix that in either OctoPrint or the libary its using.

lciscon, 2020-03-07:
> Thanks for your feedback, All your suggestions would be something for OctoPrint because that is responsible for updating the information and keeping the connection alive with its websockets. All [...]

I understand and agree in general.  However in this case the embedded browser configuration is established by TouchUI (or rather the TouchUI install scripts) not OctoPrint.  So if there is a browser setting that is incorrect it has to be changed in TouchUI.

Also keep in mind that the run-time behavior of the TouchUI environment is significantly different than in other OctoPrint installs.  Most people don't leave a browser window to OctoPrint open for 10+ hours.  And even if they did their PC would probably go to sleep and the subsequent wakeup would trigger a refresh - thus avoiding the issue.

Net net it seems like a cumulative system behavior that only manifests in TouchUI installs.  Whatever the final fix involves it has to be diagnosed first at that level.

BillyBlaze, 2020-03-08:
>I understand and agree in general. However in this case the embedded browser configuration is established by TouchUI (or rather the TouchUI install scripts) not OctoPrint. So if there is a browser setting that is incorrect it has to be changed in TouchUI.

I am not sure what you mean with browser configuration setup by TouchUI the installer; There is no such thing in the installer. However I cannot rule out that there is command line parameter like @Vertigo1206 mentioned that might trigger a race condition, but I think this would be highly unlikely. You can play with those parameters in the chromium.xinit file in the TouchUI-autostart folder.

> Also keep in mind that the run-time behavior of the TouchUI environment is significantly different than in other OctoPrint installs. Most people don't leave a browser window to OctoPrint open for [...]

Al-through I agree that a common user don't do this, I cannot rule out that people are doing this and have disabled sleep. Also you assume that a wakeup would trigger a refresh. WebSockets would be able to resubscribe to the feed without refreshing.

>Net net it seems like a cumulative system behavior that only manifests in TouchUI installs. Whatever the final fix involves it has to be diagnosed first at that level.

On first sight this seems true, however I still think this problem lies into re-subscription to WebSockets on the client-side. Since refreshing the browser will fix this issue; If there was a system wide problem (e.g. no more space to write) or an expired sessions then you wouldn't be able to do other things. My experience with failing WebSockets is typically a bug in the resubscription or the server-side timing out and not on the client. This is where I would start debugging; but since I am unable to reproduce this, I am therefor unable to find the potential issue.

If you want to fix this; I would gladly give assistance. I would start by attaching the remote debugger and wait unilll it stop refreshing data and see what errors are thrown in the console or what the status is of the WebSockets in the network tab. (e.g. does it still receive events from the server-side)

closed 2020-12-09 by github-actions[bot]