BillyBlaze/OctoPrint-TouchUI #282: I have a login screen on my 7" diplay to access Touch Ui

RayWC62, 2019-01-25:
I have followed the procedure several times and nothing works. Access control is set true with my user name. If I connect a key board to manually enter user and password that doen't work either. There is @ and # in my user name and they don't display correctly. @ displays as " and # displays as £. So I can't even sign in with a keyboard connected to the Pi.

BillyBlaze, 2019-01-25:
1. Did you mean with access control the autoLoginAs functionality?
2. You need to change your keyboard layout to your localisation (it's defaulted on GB) by running `sudo raspi-config` see https://www.raspberrypi.org/documentation/configuration/raspi-config.md

RayWC62, 2019-01-26:
I have the keyboard characters fixed. I forgot about that setting.
After the RPi boots it come to the Octoprint log in screen. I have to enter username and password every time it boots even though I check the remember me box. I have Octoprint 1.3.10

Jaxel, 2019-02-03:
How do we fix this issue? I am getting a login screen as well. Octoprint 1.3.10

bolsoncerrado, 2019-02-06:
U may wanna check https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/265

derjacks, 2019-02-19:
Same issue, tried suggestions in #265 and #266.  Watched Octoprint log and it issues an autologin but that doesn't seem to effect the browser on the touch screen.
`2019-02-18 21:31:42,219 - octoprint.server.util.flask - INFO - Passively logging in user kiosk from ::ffff:127.0.0.1 via autologin
2019-02-18 21:31:42,296 - octoprint.server.util.flask - INFO - Passively logging in user kiosk from ::ffff:127.0.0.1 via autologin
2019-02-18 21:31:42,372 - octoprint.server.util.flask - INFO - Passively logging in user kiosk from ::ffff:127.0.0.1 via autologin`

derjacks, 2019-02-19:
Just did a fresh os install and put the following in the config.yaml
          localNetworks:
          - "127.0.0.0/8"
         - "::1/128"

reboot and login screen still comes up on touch screen BUT just touching login without entering anything in either field(username or password) and it lets me in.  Odd thing is I tried that before but maybe something with the Chromium browser cache was blocking me.

derjacks, 2019-02-19:
Not stable.  After 4 reboots, wait 6 hours and 1 reboot wait 5 minutes and just touching login now does NOT work.  Reboot again and press login right away with empty fields before I guess octoprint is fully loaded and it works. I can't reproduce this pattern.  Is there a log I can look at or turn on for chromium?  Maybe the browser cache is playing games with me?

derjacks, 2019-02-19:
I did notice on earlier tries that touchui connected to port 5000 which in the documentation looked like it use to be the only port open on octoprint but now port 80 and 443 works on the browser but 5000 doesn't.  Maybe this is the issue in TouchUI setup..  Watching the touchui startup I see it connecting at bottom of the screen to 
`http://[::1]/`
which shows no port number.

closed 2019-03-12 by stale[bot]