BillyBlaze/OctoPrint-TouchUI #286: Touch UI Status Screen Freezes

TheNore, 2019-02-06:
I have noticed on 3 in 10 prints that the main status screen of touch UI will "freeze" on any status information mid print, or at the end of the print. It will not refresh on an initial print load when the latter happens. The screen and menus are still useable, and a manual TouchUI restart clears up the problem.

TheNore, 2019-02-14:
[image]

Jaxel, 2019-02-25:
I have the same issue.

TheNore, 2019-03-11:
This still happens, but no fixes or remedies available?

BillyBlaze, 2019-03-11:
Sorry for the long delay and thanks for reporting the issue!

Since we have two open issues about the same issue, I propose to continue this in #281, also could you please share your `octoprint.log` in that ticket too?

closed 2019-03-11 by BillyBlaze