BillyBlaze/OctoPrint-TouchUI #292: Auto-login for TouchUI

jglazer63, 2019-03-03:
Is this feature going to be fixed?   I've just installed TouchUI for the first time about a month ago figuring that it would be corrected at some point but it doesn't look like anyone is trying to fix it.  Maybe I am missing something myself?

Thanks

BillyBlaze, 2019-03-04:
Hi @jglazer63,

The Auto-Login is OctoPrint functionality and not TouchUI. I can help figuring out if the config is wrong, but beside that I cannot fix anything on my side.

There are some upcoming changes in OctoPrint 1.3.11 (e.g. More logs about Auto-Login and default IPv6 support)

closed 2019-03-25 by stale[bot]

evanquinnalaska, 2019-04-04:
I just updated my octoprint to the current RC and it fixed this problem for me. Fyi