BillyBlaze/OctoPrint-TouchUI #293: Broken with Waveshare 3.5 inch SPI LCD

Jozer99, 2019-03-07:
I have a Raspberry Pi 2 running Octoprint 0.16.0 with a Waveshare 3.5 inch SPI LCD.  A year ago, I had this setup running your boot-to-chrome plugin just fine, but retired the system when I got a Raspberry Pi 3 and larger LCD.  

I tried setting it up again to use with a second printer.  I got the Waveshare drivers installed fine, and I can see the scroll of boot-up text.  But when boot-to-chrome starts, I see a blank screen with a cursor at the top left.  Occasionally I see a flash of the TouchUI window, but it is quickly replaced with the black screen again.  Touching the screen seems to prompt additional brief flashes of TouchUI.  When I tweak the X configuration to use fb1, I just see the blank screen with no flashes of TouchUI at all.  

What do I have configured incorrectly?  I see a couple of other similar sounding problems in the issue tracker, but they were closed out without any details being provided on the fix.

Jozer99, 2019-03-09:
I resolved the issue.  If anyone else has the problem, the solution is to install the waveshare LCD using the 
"lite" modifier.  For instance, for the 3.5 inch waveshare, use the command 

LCD35-show lite

When you install the LCD.  This is necessarily because Octopi is built on the Stretch Lite version of Raspbian.  If you have already installed the LCD incorrectly, just go back into the LCD-show directory and rerun the correct command, then then reboot.

closed 2019-03-09 by Jozer99

n3bojs4, 2019-05-01:
Thanks a lot @Jozer99 you fixed my issue :)