BillyBlaze/OctoPrint-TouchUI #298: X11 doesn't seem to read the 99-calibration.conf file

SteveDallas, 2019-04-23:


BillyBlaze, 2019-04-23:
Try moving the file to this directory instead: `/etc/X11/xorg.conf.d/`

SteveDallas, 2019-04-23:
I tried that, too.  It didn't seem to work there, either.

BillyBlaze, 2019-04-23:
Have you tried setting `Option "InvertY" "true"` too?

BillyBlaze, 2019-04-23:
see https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/54

SteveDallas, 2019-04-26:
I got things working by doing a complete reinstall and using Waveshare's LCD-show repository (https://github.com/waveshare/LCD-show) to install the driver, rather than https://github.com/goodtft/LCD-show

When reinstalling TouchUI on the stretch lite based Octopi 0.16.0, I found that the configuration files for X11 were in /usr/share/X11/xorg.conf.d/ rather than /etc/X11/xorg.conf.d/, so when I got the "boot to black screen" issue, I had to adjust my procedure from your troubleshooting instructions.

Everything is working now and I can install the rest of my plugins.

closed 2019-05-15 by BillyBlaze