BillyBlaze/OctoPrint-TouchUI #299: touch UI setting not open, and mobile not detect

devildant, 2019-05-04:
Hello,
I am under OctoPrint 1.3.11rc3, I can not open touch settings, and mobile detection does not work.
I do not have an error in the console

jneilliii, 2019-05-05:
I can confirm this same thing in 1.3.11rc3 in Chrome. Attempting a force of touchUI using `#touch` at the end of the url doesn't force touchui to load either.

BillyBlaze, 2019-05-15:
Thanks for reporting! I can reproduce this issue and I'll fix it in the upcoming release.

Rapsey, 2019-05-15:
I was experiencing the same problem. Just tried installing from the maintenance branch and I can confirm this has solved the problem.

Probably not related to this issue but it might be worth pointing out that when doing this I get 3 TouchUI entries in the OcroPrint settings menu, of which only the third one works. I guess this might be because the maintenance branch isn't meant to be release-ready which is why I mention it here instead of creating an issue.

Thank you for fixing!

Drzet, 2019-05-25:
Touch/mobile detection still doesn't work with the newest version. The mobile icon at the top doesn't work either

closed 2019-05-25 by BillyBlaze

BillyBlaze, 2019-05-25:
Thanks again for reporting!

This has been released in 0.3.13

Jan07576, 2019-07-14:
I just had similar issues with the latest version that I downloaded today.

commit e9355524 (2019-05-15): :bug: Fix compatibility with OctoPrint 1.3.11

Fixes #299