BillyBlaze/OctoPrint-TouchUI #302: TouchUI stops working when Shutdown Printer Plugin is active

ARCH-3D, 2019-05-27:
Hello,

After installing the last upgrade the TouchUI stops workinkg. And the problem is with Shutdown Printer plugin. When the ShutDown Printer plugin is active the TouchUI stops working and when is disabled the TouchUI works properly. I would like to know if you could solve this incompatibility. Many thanks.

Best regards,

BillyBlaze, 2019-05-27:
Hello,

Thanks for reporting. I can reproduce your issue and will fix this in the upcoming release.

BillyBlaze, 2019-05-27:
This has been fixed on the maintenance branch and will be released in the next version. There is no E.T.A. but if you want this you could install TouchUI from maintenance.

Install with this link: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

ARCH-3D, 2019-05-27:
sorry for the question. But how can I install with this link: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip ?

ARCH-3D, 2019-05-27:
And many thanks for your help :)

ARCH-3D, 2019-05-27:
Many thanks BillyBlaze :) It works now! Great Work ! 👍

Kuenstler79, 2019-08-19:
How can I install a zip file from the maintenance branch? Unfortunately I have only little linux experience.

Varming73, 2020-01-25:
Im on TouchUI: 0.3.13, OctoPrint: 1.3.12  and Shutdown Printer (1.0.8) and the problem is still there. What am I missing?

commit 22d2023b (2019-05-27): :bug: Fix Shutdown Printer Plugin

While disabling Themify the Shutdown Printer Plugin no longer works

Fixes #302
closed 2020-03-11 by BillyBlaze