BillyBlaze/OctoPrint-TouchUI #303: Major issues with release 0.3.13

FrankX0, 2019-05-28:
I have major issues with the latest TouchUI release.
I have been using TouchUI in the last few months with great success.
But after updating a few days ago, I suffer from following (and more) defects:

- Status of the printer or job is not shown anymore
- The file list is empty
- Cannot see the webcam anymore ("Webcam stream not loaded")
- Missing menus
- Etc.

Restarting OctoPrint, rebooting the Raspberry Pi, re-installing TouchUI: nothing helped.
Same issue on multiple platforms. Switching back to the normal OctoPrint interface on the same platform results in normal behaviour.

Is there anyone else facing this? Any resolution?

BillyBlaze, 2019-05-29:
Hello,

Thanks for reporting, what plugins do you have installed?

FrankX0, 2019-05-29:
Hi Paul,

This is the complete list:
- Action Command
- Prompt Support  
- Announcement Plugin  
- Anonymous Usage Tracking  
- Application Keys Plugin  
- Automatic Shutdown (0.1.4)
- Backup & Restore  
- Core Wizard  
- Discovery  
- Error Tracking  
- Firmware Updater (1.4.0)
- Force Login  
- Fullscreen Plugin (0.0.4)
- Logging  
- MQTT (0.8.2)
- Navbar Temperature Plugin (0.11)
- Octolapse (0.3.4)
- Pi Support Plugin  
- Plugin Manager  
- PortLister (0.1.8)
- Printer Safety Check  
- Simple Emergency Stop (0.2.5)
- Software Update  
- Tasmota-MQTT (0.3.1)
- The Spaghetti Detective (Beta) (0.6.0)
- Themeify (1.2.0)
- TouchUI (0.3.13)
- Virtual Printer

BillyBlaze, 2019-05-29:
Hi @FrankX0,

Your issue seems exactly the same as #302 , try to install TouchUI with this link to see if this also fixed your issue: https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip

FrankX0, 2019-05-29:
Yes, same issue.
Thanks for your support.

closed 2019-05-29 by FrankX0