BillyBlaze/OctoPrint-TouchUI #304: console error in 0.3.13 cause observable pb

devildant, 2019-05-29:
```
packed_libs.js?ea9fbc9a:3819 jQuery.Deferred exception: obj[2].indexOf is not a function TypeError: obj[2].indexOf is not a function
    at 192.168.1.18/static/webassets/packed_plugins.js?a80e3ab6:10811:32
    at Function.Nn.remove (192.168.1.18/static/webassets/packed_libs.js?ea9fbc9a:11267:432)
    at TouchUI.plugins.themify (192.168.1.18/static/webassets/packed_plugins.js?a80e3ab6:10810:12)
    at Function.moment.locale (192.168.1.18/static/webassets/packed_plugins.js?a80e3ab6:10061:36)
    at HTMLDocument.<anonymous> (192.168.1.18/static/webassets/packed_core.js?79f3153d:15465:16)
    at mightThrow (192.168.1.18/static/webassets/packed_libs.js?ea9fbc9a:3535:29)
    at process (192.168.1.18/static/webassets/packed_libs.js?ea9fbc9a:3603:12) undefined
jQuery.Deferred.exceptionHook @ packed_libs.js?ea9fbc9a:3819
process @ packed_libs.js?ea9fbc9a:3607
setTimeout (async)
(anonymous) @ packed_libs.js?ea9fbc9a:3641
fire @ packed_libs.js?ea9fbc9a:3269
fireWith @ packed_libs.js?ea9fbc9a:3399
fire @ packed_libs.js?ea9fbc9a:3407
fire @ packed_libs.js?ea9fbc9a:3269
fireWith @ packed_libs.js?ea9fbc9a:3399
ready @ packed_libs.js?ea9fbc9a:3879
completed @ packed_libs.js?ea9fbc9a:3889
?time=1559148890:10684 Initialized error tracking
packed_core.js?79f3153d:11012 Uncaught TypeError: obj[2].indexOf is not a function
    at packed_plugins.js?a80e3ab6:10811
    at Function.Nn.remove (packed_libs.js?ea9fbc9a:11267)
    at TouchUI.plugins.themify (packed_plugins.js?a80e3ab6:10810)
    at Function.moment.locale (packed_plugins.js?a80e3ab6:10061)
    at HTMLDocument.<anonymous> (packed_core.js?79f3153d:15465)
    at mightThrow (packed_libs.js?ea9fbc9a:3535)
    at process (packed_libs.js?ea9fbc9a:3603)
(anonymous) @ packed_plugins.js?a80e3ab6:10811
Nn.remove @ packed_libs.js?ea9fbc9a:11267
TouchUI.plugins.themify @ packed_plugins.js?a80e3ab6:10810
moment.locale @ packed_plugins.js?a80e3ab6:10061
(anonymous) @ packed_core.js?79f3153d:15465
mightThrow @ packed_libs.js?ea9fbc9a:3535
process @ packed_libs.js?ea9fbc9a:3603
setTimeout (async)
(anonymous) @ packed_core.js?79f3153d:11012
jQuery.readyException @ packed_libs.js?ea9fbc9a:3827
(anonymous) @ packed_libs.js?ea9fbc9a:3847
mightThrow @ packed_libs.js?ea9fbc9a:3535
process @ packed_libs.js?ea9fbc9a:3603
setTimeout (async)
(anonymous) @ packed_libs.js?ea9fbc9a:3641
fire @ packed_libs.js?ea9fbc9a:3269
fireWith @ packed_libs.js?ea9fbc9a:3399
fire @ packed_libs.js?ea9fbc9a:3407
fire @ packed_libs.js?ea9fbc9a:3269
fireWith @ packed_libs.js?ea9fbc9a:3399
process @ packed_libs.js?ea9fbc9a:3623
setTimeout (async)
(anonymous) @ packed_libs.js?ea9fbc9a:3641
fire @ packed_libs.js?ea9fbc9a:3269
fireWith @ packed_libs.js?ea9fbc9a:3399
fire @ packed_libs.js?ea9fbc9a:3407
fire @ packed_libs.js?ea9fbc9a:3269
fireWith @ packed_libs.js?ea9fbc9a:3399
ready @ packed_libs.js?ea9fbc9a:3879
completed @ packed_libs.js?ea9fbc9a:3889
```
```

 TouchUI.prototype.plugins.themify = function() {
        
        	_.remove(OCTOPRINT_VIEWMODELS, function(obj) {
        		if (obj[2] && obj[2].indexOf("#settings_plugin_themeify") !== -1) {
        			console.info("TouchUI: Themeify is disabled while TouchUI is active.");
        			return true;
        		}
        		
        		return false;
        	});
        
        }
```

devildant, 2019-05-29:
cause pb, with tplink plugin, and your own plugin (no menu touchui), and with login button
[image]

BillyBlaze, 2019-05-29:
hello @devildant, thanks for reporting, this issue has already been fixed on the maintenance branch (see https://github.com/BillyBlaze/OctoPrint-TouchUI/commit/22d2023bd3fb20e23afe40bb1050f24f347bf58e)

devildant, 2019-05-29:
hooo ok thank you
I did not see the maintenance branch: p
(I adapted my plugin "shutdown printer "and my future plugins so that they work with touchui, icon ....)

closed 2019-06-19 by stale[bot]