BillyBlaze/OctoPrint-TouchUI #305: Unable to boot to browser.

Capp3, 2019-06-04:
I have gone through other issues, but can't get boot to browser, on a 3.2" TFT, working on a R-Pi 3. I have tried user "pi" and "root". I have also tried both screen addresses "fb1" and "fb0"

here is the output of 

> sudo xinit ~/TouchUI-autostart/chromium.xinit

```
X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=f316bd05-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait fbcon=map:10 fbcon=font:ProFont6x11
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support) 
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Jun  4 16:11:59 2019
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
sed: can't read /run/touchui-chromium//Default/Preferences: No such file or directory
sed: can't read /run/touchui-chromium//Default/Preferences: No such file or directory
sed: can't read /run/touchui-chromium//Default/Preferences: No such file or directory
sed: can't read /run/touchui-chromium//Default/Preferences: No such file or directory
TouchUI: OctoPi detected
TouchUI: haproxy has a listening IPv6 socket.
TouchUI: touch events automatically detected
TouchUI: screensaver enabled
python: can't open file '/home/root/TouchUI-autostart/helpers/server.py': [Errno 2] No such file or directory
wm_set_cursor_visibility: visible using XCreateFontCursor
[1333:1333:0604/161201.417515:ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
```

BillyBlaze, 2019-06-08:
Hi @Capp3,

Can you run this as user PI and post the output?

Capp3, 2019-06-12:
To be honest I got impatient and re-flashed my card and went through the process again. 

I'm guessing I either screwed something up, or skipped a step. 

Thank you for the follow up.

closed 2019-06-12 by Capp3