BillyBlaze/OctoPrint-TouchUI #307: Failed installed.

tedukes, 2019-07-02:
I spent about 4 hours last night trying install TouchUI. Checked, double checked, triple checked the steps to install the drivers and TouchIU. All I could get was a black screen flashing TouchUI. Couldn't access Octoprint via the web as something caused it to crash. Never could get Octoprint to start. I uninstalled the TouchUI bootloaded. Still couldn't get Octoprint to start. 

I had to flash a new image of Octoprint to get it working.

I have been using Red Hat/RHEL since 1997. Something is broken.

Lissanro, 2019-07-09:
I just installed TouchUI and everything works for me. I suggest that you provide more details, such as what version of Octoprint you were using and where it was installed (for example, I have Raspberry Pi 3 B+ with OctoPi 0.16.0). You also did not  if you looked at octoprint.log - perhaps it contained relevant information why OctoPrint could not start and if it was really related to TouchUI. You also did not mention if the issue is still reproducible for you after installing new OctoPrint.

tedukes, 2019-07-09:
I have up and running now. Using the current version of Octoprint and a RPi3 B+. The log quit when Octoprint quit.
 I was using the filament runout reloaded plugin that uses pins 18 and 20. I didn't think about un-installing prior. Maybe that was the issue, IDK.

Followed this guide, had to tweak some orientation settings, but its working.
https://pastebin.com/rsMa5K1v

closed 2019-07-30 by stale[bot]