BillyBlaze/OctoPrint-TouchUI #309: LCD shown Login window instead of TouchUI

athulsnair, 2019-07-09:
Hello, 
  After installing the Touch UI, the LCD still shows the login screen. I have tried every step from start(Including Octoprint installation) but it doe
[image]
sn't show TouchUI.
In my first try I have updated the Octoprint to latest version, since it didn't work I have went with the old one i had (Octoprint 1.3.8 running on Octopi 0.15.1), which is also what used in the [video ](https://www.youtube.com/watch?v=rqy3dVZ4JH4)that I'm using for reference.

I had trouble during the driver installation. It was showing like some packages were not installed, so I tried to manually install it. Which installing one of those install, it had some problem and recommended to run _**sudo apt --fix-broken install**_ , after that I was able to install the driver, although it still showing letters in small size, instead of what says in the video.

What could be the reason for this?

Thanks

athulsnair, 2019-07-09:
After reading other issue
**_`sudo xinit ~/TouchUI-autostart/chromium.xinit`_**

```
pi@octopi:~ $ sudo xinit ~/TouchUI-autostart/chromium.xinit
[sudo] password for pi:

X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Jul  9 11:48:30 2019
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.

```

Also did as the guide for "[Boot to Black screen](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Troubleshooting#boots-to-blackscreen)" 
The file wasn't there. i had to create it
Currently it's in fb0, tried fb1 as well

```
Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb0"
EndSection
```

BillyBlaze, 2019-07-09:
Please remove `/etc/X11/xorg.conf.d/99-fbdev.conf` and change the /fb0 or /fb1 in the file generated under `/usr/share/X11/xorg.conf.d/99-fbdev.conf` and rerun `sudo xinit ~/TouchUI-autostart/chromium.xinit`

athulsnair, 2019-07-09:
There was no file named 99-fbdev.conf, so i created on and pasted this

```
Section "Device"
  Identifier "touchscreen"
  Driver "fbdev"
  Option "fbdev" "/dev/fb0"
EndSection
```
and this is the output after running `sudo xinit ~/TouchUI-autostart/chromium.xinit`

```
pi@octopi:~ $ sudo xinit ~/TouchUI-autostart/chromium.xinit

(EE)
Fatal server error:
(EE) Server is already active for display 0
        If this server is no longer running, remove /tmp/.X0-lock
        and start again.
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE)
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 7 requests (7 known processed) with 0 events remaining.
pi@octopi:~ $

```

LCD is displaying TouchUI screen now. I haven't connected Raspberry Pi to printer now, need to check that

Why wan't the file created first time?

BillyBlaze, 2019-07-09:
Did you install it manually or with the install helper?

athulsnair, 2019-07-25:
I use this [video ](https://www.youtube.com/watch?v=rqy3dVZ4JH4)as reference. Downloaded the plugin from octoprint and then using the guides as per your github page [](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser)

closed 2019-08-15 by stale[bot]