BillyBlaze/OctoPrint-TouchUI #311: Font set to incorrect value, screen seems blank

Jeff-Haas, 2019-07-11:
Please see the screenshot - the yellow star is from the Print Time Genius plugin.  Otherwise the info seems to be blank.

[image]

However if I highlight the text you can see it's there:

[image]

Jeff-Haas, 2019-07-11:
This is running on a Google Pixel 2, with the current version of Android Chrome.  Disabling Print Time Genius makes no difference - you just don't see the little yellow star!

Jeff-Haas, 2019-07-19:
I was able to figure out how to change the font color so that it shows up for me, but it seems that the default is incorrect for the current Android Chrome browser.

celogeek, 2020-01-04:
To fix this I create a small less file:

```less
#printer {
    .accordion {
        &-inner {
            color: @main-background;
        }
    }
}
```

So I can add it to customization, use local file (copy it to your raspberry).

closed 2020-03-11 by BillyBlaze