BillyBlaze/OctoPrint-TouchUI #312: Toggled to Touch UI - can't toggle back

Jan07576, 2019-07-14:
After switching view to Touch UI once it seems impossible to toggle back to regular. Also some buttons seem incomplete; I can not access the settings anymore.
[image]

freddotu, 2019-07-14:
As a newbie to TouchUI, I was not aware there was a feature to permit toggling. Is this a screen touch location or otherwise?

Jan07576, 2019-07-14:
After installing I noticed that it won´t go to Touch UI mode on my mobile phone - so I activated "Toggle Touch UI" in the user settings. Still no effect. When I noticed a smartphone - icon in my regular web-surfece i was able to activate Touch UI manually. Now there does not seen to be a way back.

Jan07576, 2019-07-14:
Does anyone know how to access the settings by adress-line (soemthing like http://192.168.178.63/**#control**)  there has to be a way to remove this messy plugin even with the button missing.

BillyBlaze, 2019-07-14:
Hi @Jan07576,

Thanks for reporting. There seems to be a javascript error. Is there a way you can share your browser log?

BillyBlaze, 2019-07-14:
Hi @freddotu,

You can toggle TouchUI in the [TouchUI user-settings](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Configuration#user-settings) modal and should open if you click the 'mobile' icon,

freddotu, 2019-07-14:
My ignorance may be getting in the way. I installed from an image downloaded from the Octoprint web site. During the configuration, I selected to automatically start Octoprint and TouchUI plug in was added shortly after that.

I have version 1.3.11, although the github page shows 0.3.11 as the current version.

The Pi is connected to the printer and the Pi also has the touchscreen interface. It's a healthy 7" screen. I've been wanting on occasion to exit the interface, to exit Octoprint to get to a linux terminal screen at the very least but not sure how to accomplish that and then how to return to Octoprint/TouchUI.

[image]

I don't have the toggle suggested in the link provided, nor some of the other references in that link.

Am I heading in the wrong direction?

BillyBlaze, 2019-07-14:
No, your not heading in the wrong direction. This option is disabled on touchscreen devices that use the bootloader.

freddotu, 2019-07-14:
Is there a method to exit to a terminal (linux) and restart later, using the touchscreen?

JSViterise, 2019-07-19:
hey @BillyBlaze I am having these same difficulties. touch UI interface on all my devices, touch screen, web browser and mobile phone with no printer detection, or option to toggle back to regular octoprint. octoprint is completely unresponsive. I have no idea what to do from here?

Jan07576, 2019-07-19:
Sorry, I was not able to retrieve a log from my browser, also I gave up and installed Octoprint from scratch, this time without TouchUI.

klinkenberg, 2019-07-29:
Hallo,
I have the same problem, how to reach X or a terminal temporarly from touchUI (fullscreen boot) ?

daniel11krupa, 2019-07-29:
erase cookies from to your browser

klinkenberg, 2019-07-31:
How to erase cookies if I can not get a teminal? (and no keyboard at hand ....)

closed 2019-08-22 by stale[bot]

galosre, 2020-01-01:
Sorry to reopen this,but when I toggle Touch UI no way I can go back to normal view,  my  only option is to delete octoprint cookies
So where you are in touch UI where is the icon to toggle?