BillyBlaze/OctoPrint-TouchUI #316: Installing dependencies Failed

Alexthefox, 2019-07-25:
Hi, i cant install touchui with my display. I try from 3 days, i have try different installation metod, but cant have the last step.
this is my procedure:
fresh Octoprint installation
Install TFT driver https://github.com/goodtft/LCD-show.git and follow vendor istruction
install TouchUi from Octoprint webpage (also try by command line)
Install https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
after, sudo ~/TouchUI-autostart/helpers/install

Running apt-get update (this will take a while)
Do NOT switch off the Pi or close this console until done!
 - success

Installing dependencies (xinit xinput xserver-xorg xserver-xorg-video-fbdev x11-xserver-utils matchbox unclutter chromium-browser):
 - failed

in this last step, i get  this 
[image]

any help?

tgmorris99, 2019-07-25:
I was coming here to post the exact same thing.

Install latest octopi
Change config.txt to support the LCD display
sudo raspi-config

reboot

Update OctoPrint to 1.3.11
Install TouchUI Plugin

git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
sudo ~/TouchUI-autostart/helpers/install

```
 Install helper v0.0.2
 
 Boot to command line:
  - success
 
 Running apt-get update (this will take a while)
 Do NOT switch off the Pi or close this console until done!
  - failed
 ```

I've tried with and w/o install of the LCD driver which expects the X11 environment to already be there so it partially fails.

There's no log file in /var/log/xxxxxx so it's hard to troubleshoot.
Is there a clear set of steps to execute manually?

BillyBlaze, 2019-07-26:
Thanks for reporting, can you run `sudo ~/TouchUI-autostart/helpers/install --debug`  for some more details

tgmorris99, 2019-07-26:
That helped. Seems that the mirror was slow as the speeds would drop to nonexistent and resulted in a few Hash Sum Mismatch issues. so I ran
`sudo apt-get update --fix-missing`
and then reran the install script with the debug option. Kept doing this until everything was downloaded and the install successfully completed.

closed 2019-08-16 by stale[bot]