BillyBlaze/OctoPrint-TouchUI #318: TouchUI Broken

juacuell, 2019-07-27:
TouchUI on Pi 3b broken after updating to latest octoprint. When I disabled that plugin the Pi will show desktop version fine. Have tried reinstalling desktop environment and still no luck

Kuenstler79, 2019-07-29:
I have the same problem.

chrschn, 2019-08-17:
Same here, TouchUI on OctoPrint v1.3.11 loads on my phone but no controls are available after login. The connections dialog is empty. File list, temperature, and control windows are mostly empty.
[image]
[image]
[image]

BillyBlaze, 2019-08-17:
Thanks for reporting, can you provide me a list of plugins? If you're runnig Shutdown Printer Plugin, then there is already fix (see #302) on the maintenance branch.

chrschn, 2019-08-17:
Thanks for the fast response. I can confirm that I have the Shutdown Printer plugin installed. I disabled it and upon restart of OctoPrint, the TouchUI works again. Thanks for the pointer, I'm looking forward for the fix in the next release.

Also, thanks for creating TouchUI! It works really well and makes it so easy to navigate OctoPrint on my phone. I love it!

Kuenstler79, 2019-08-19:
Thank You for the hint! That worked for me too.

closed 2019-09-09 by stale[bot]