BillyBlaze/OctoPrint-TouchUI #321: Support for very small (480x320) screens

steviehs, 2019-08-02:
I would love to have a better looking ui on my small 480x320 LCD added to the Pi, but touchui in the default settings does not look very well there. Maybe someone has already worked on that?

Kiron21, 2019-08-14:
Yes, especially open file dialogue (touuched by mistake) is way to reboot TouchUI wia ssh or all octprint - no way to close it (buttons out off screen).

Erik262, 2019-11-28:
As I heard about the TouchUI plugin for Octoprint (Youtube etc.) I was very happy. Because I could now replace the display from my printer with these from my pi. By this I can also update to Marlin 1.1.9 which would no longer work with my Original display, but who cares if I can get Octoprint with TouchUI. 

Now I have this sad picture:
[image]

Just making the icons or something else smaller would be good enough for any 3.5 inch Display (480x320)

yewtink, 2020-01-29:
I am having a similar issue using [https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=9&cad=rja&uact=8&ved=2ahUKEwjlkqDR36fnAhXKmXIEHXJCBvIQFjAIegQIAhAB&url=https%3A%2F%2Fwww.uctronics.com%2Fdownload%2FAmazon%2FB0106.pdf&usg=AOvVaw0__0RxwQW8RdGBHjHn4sVf](url)

I was able to adjust the resolution by editing the  /boot/config.txt 

> hdmi_cvt 800 533 60 6 0 0 0

It has improved but the touchscreen calibration is way off still.

BillyBlaze, 2020-03-08:
Besides the file browser; What are the common pitfalls on this resolution? I will not write a new interface for this resolution but I can tweak it.

geoffdavis, 2020-04-22:
@BillyBlaze Gcode viewer doesn't stay on the same screen.

Header for the menu on all screens takes a lot of horizontal real estate - may be better on left side.

Current file window doesn't fit on one screen due to above menu issue.

closed 2020-12-09 by github-actions[bot]