BillyBlaze/OctoPrint-TouchUI #326: TouchUI loads on Raspberry Pi used as desktop

xh3dx, 2019-09-20:
I have a raspberry pi, used as a desktop, running full raspbian in my garage next to my 3d printer. When ever i pull up the octoprint site for my printer, it loads TouchUI automatically, which i don't want it to. When i go to the touch UI settings, there is no option to "Toggle TouchUI." Is there a fix for this?

I have tried by disabling both "Start TouchUI Automatically" options in the settings but this doesn't help.

spiff72, 2019-10-02:
Chiming in with the same issue.  Running on a 1920x1080 monitor, chromium as the browser, on a Raspi 4 with 4GB RAM.  Can't for the life of me get rid of the touch UI interface on this machine.  Other machines it works fine.

spiff72, 2019-10-02:
One additional comment - while I don't have a "Toggle TouchUI" option, I do have a Toggle Touchscreen Mode option (which doesn't help).  I also turned off all of the options that should start it automatically (based on resolution and touchscreen being present).

spiff72, 2019-10-02:
And one LAST clarification!  The Pi that I am running this on is NOT the pi that is running my Octoprint server.  Totally separate machine, but on the same network, of course...

Thanks!

spiff72, 2019-10-17:
Bumping this, as it is still an issue for me...

xh3dx, 2019-10-17:
Are we to expect no response from the developer? I still have this issue. Raspberry Pi 4 (4GB)

BillyBlaze, 2019-10-18:
Sorry for the long delay, the problem is caused by this [commit](https://github.com/BillyBlaze/OctoPrint-TouchUI/commit/d67b8a8467556661eda5868f035ae7cfe17aedac) and the message of this fix is:

> You could disable TouchUI while using the bootloader, therefor the bootloader would fail to start because 'OctoPrint loaded without TouchUI'. Therefor always load TouchUI based on the user-agent string we custom set in chromium.xinit

Anyhow I will fix this in the next release.

commit dafca68b (2020-03-07): :bug: Don't show TouchUI on Raspian desktop

Previously we would always load TouchUI if the user-agent was
a chromium browser running in X11. Now we only force load TouchUI
if the user-agent contains 'TouchUI'

Fixes #326
closed 2020-03-11 by BillyBlaze