BillyBlaze/OctoPrint-TouchUI #334: Make cancel print dialog box smaller

purist69, 2019-10-15:
I run TouchUI on a 800x480 touchscreen.  When I press stop on a print job a confirmation dialog comes up that is too large and the button appears off the screen.  Please resolve.

BillyBlaze, 2019-10-18:
Thanks for reporting, I'll have a look at this for the next release.

ellindsey, 2019-10-28:
I may be having this issue as well.  When confirmation dialogs come up, the UI will frequently ignore me pressing "yes" and act as if I pressed "no".

compmas2, 2020-01-10:
I have been seeing the same issue. This is not just the print job cancel confirmation screen, but any confirmation screen within TouchUI when using a 800x480 touchscreen.

Additionally on the 800x480 touchscreen, the three horizontal lines which is supposed to bring up more plugins or menus is sometimes hijacked by the tab icon behind it and that is pressed instead of the three horizontal lines icon.

BillyBlaze, 2020-03-07:
Closing this issue as its duplicate of #296 
For updates follow that issue thread

closed 2020-03-07 by BillyBlaze