BillyBlaze/OctoPrint-TouchUI #335: Octopi with TouchUI - not loading

mcole79, 2019-10-19:
Hi all,

I appear to have what has been a common issue with TouchUI not being able to log in when using http://[:1]/. It just times out to red screen.

I've read all the issues and tried everything i've found to try and resolve, but alas i've still not got it working. Can anyone offer any advice on this, or is it a known bug which cannot be resolved?

I'm fairly new to Pi and Octopi/Octoprint, so theres a very good chance i'm just missing something obvious.

Regards
Martin

wesselah, 2019-10-27:
I am having excactly the same issue but i even don't have the helpers directory on /TouchUI-autostart mayby i missed some?

wesselah, 2019-10-27:
got it resolved i did a new git pull and now i have the helpers directory

NoNaym, 2019-11-09:
I have been fighting with what I believe to be the same issue as well, and have come to the conclusion that the drivers for the Waveshare 5inch HDMI LCD that I am using are causing the issue. I did a clean OctoPi install followed by TouchUI (and autostart) and fixes for current OctoPrint versions, and found that the TouchUI GUI was able to load up perfectly... then I installed the touchscreen driver, and BAM! My touchscreen starts to work, but now TouchUI won't load (it tries, but ends in Reds Screen of Death).

Unfortunately I do not have a fix for you right now, but hopefully @BillyBlaze might be able to revisit this when he has some time. Hoping I am on to a potential cause of this issue, just looking to Billy for a fix! :)

closed 2019-12-01 by stale[bot]