BillyBlaze/OctoPrint-TouchUI #338: Glitch with a new option in OctoPrint 1.3.12

yet-another-average-joe, 2019-10-29:
Hi,

I just installed OctoPrint 1.3.12. It came with a great option I asked for a while ago on OctoPrint Git : "Close dialogs by clicking outside" can be deactivated. Deactivating this feature makes use of TouchUI on inaccurate touchsreens much easier.

see here : https://github.com/foosel/OctoPrint/issues/3162

But in some circumstances, TouchUI does not like it.

Steps to reproduce :

- update OctoPrint to 1.3.12
- in Settings/Appearence, uncheck "Close dialogs by clicking outside"
- open "Settings" in the main menu, the dialog box opens quite slowly... Click anywhere while it is not 100% displayed : the edit field "Additional serial ports" will be over the menu, and can't be used.
- workaround : be carefull not to click anywhere until the Settings dialog is displayed !

The UI updates slowly (RaspBerry pi 3B+ is not an Intel I9 with the last high end GPU !). The user could think the touch event was not received, and click a second time -> glitch !

I don't know if it's a TouchUI issue, or an OctoPrint one. When "Close dialogs by clicking outside" is deactivated, and the screen is touched/clicked outside, there's a visual effect that could explain the problem (OctoPrint shakes the dialog horizontally).

[image]

If you'd prefer, I could open an issue on OctoPrint Git...

BillyBlaze, 2020-02-01:
Sorry for the long delay and thanks for reporting! I'll have a look at this in the next release.

BillyBlaze, 2020-03-07:
Thanks for the detailed report. I was easily able to identify the issue! This will be fixed in the upcoming release.

commit 3875547e (2020-03-07): :bug: Improve shake modal

The shake animation uses translate and interferes with
iScroll. Therefor override the animation with margin-left

Fixes #338
closed 2020-03-11 by BillyBlaze